gnea/grbl #705: ENABLE_SAFETY_DOOR_INPUT_PIN not doing what it's documented to do?

bdurbrow, 2019-08-10:
From config.h:

```
// This option causes the feed hold input to act as a safety door switch. A safety door, when triggered,
// immediately forces a feed hold and then safely de-energizes the machine. Resuming is blocked until
// the safety door is re-engaged. When it is, Grbl will re-energize the machine and then resume on the
// previous tool path, as if nothing happened.
// #define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.
```

OK, but in the ISR in system.c it seems to be disabling the feed hold input pin, and enabling the safety door input pin?

```
    #ifndef ENABLE_SAFETY_DOOR_INPUT_PIN
      if (bit_istrue(pin,CONTROL_PIN_INDEX_FEED_HOLD)) {
        bit_true(sys_rt_exec_state, EXEC_FEED_HOLD);
    #else
      if (bit_istrue(pin,CONTROL_PIN_INDEX_SAFETY_DOOR)) {
        bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
    #endif
    }
```

It seems to me that in order to follow the documentation in config.h; it should read something like:

```
    if (bit_istrue(pin, CONTROL_PIN_INDEX_SAFETY_DOOR))
      bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
    
    #ifdef ENABLE_SAFETY_DOOR_INPUT_PIN
      if (bit_istrue(pin, CONTROL_PIN_INDEX_FEED_HOLD))
        bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
    #else
      if (bit_istrue(pin, CONTROL_PIN_INDEX_FEED_HOLD))
        bit_true(sys_rt_exec_state, EXEC_FEED_HOLD);
    #endif
```

So...
* Always do the safety door pin check.
* If ENABLE_SAFETY_DOOR_INPUT_PIN is defined, map the feed hold pin to safety door behavior.
* Otherwise map the feed hold pin to feed hold behavior.

Also; based on the documented behavior of ENABLE_SAFETY_DOOR_INPUT_PIN; it seems poorly named... perhaps it should be something like `MAP_FEED_HOLD_INPUT_TO_SAFETY_DOOR_BEHAVIOR`?

Did that make sense?

chamnit, 2019-08-10:
You missed the n in #ifndef.

closed 2019-08-10 by chamnit

bdurbrow, 2019-08-10:
No, I didn't miss the 'n'. FWIW, I used a regular #ifdef instead of an #ifndef because I thought it was clearer to someone reading the source code.

What I _did_ miss is that on the '328 version you are making the assumption (or so I gather from cpu_map.h) that CONTROL_FEED_HOLD_BIT and CONTROL_SAFETY_DOOR_BIT will always be the same.

On mega, they aren't... and considering that they are in the configuration headers as two separate items, it's not unreasonable for someone to try to map them separately... so I think the main body of the code should handle that case correctly (this can be done with the preprocessor; so it shouldn't have an effect on the flash footprint for the '328 version.

BTW, the official distribution of grbl-mega still has the broken ISR that was fixed in https://github.com/gnea/grbl/issues/543

Considering the above... perhaps something more like this would be applicable:

```    
    #ifdef MAP_FEED_HOLD_INPUT_TO_SAFETY_DOOR_BEHAVIOR
      if (bit_istrue(pin, CONTROL_PIN_INDEX_FEED_HOLD))
        bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
    #else
      if (bit_istrue(pin, CONTROL_PIN_INDEX_FEED_HOLD))
        bit_true(sys_rt_exec_state, EXEC_FEED_HOLD);
      
      #ifdef SEPERATE_CONTROL_SAFETY_DOOR_PIN
        if (bit_istrue(pin, CONTROL_PIN_INDEX_SAFETY_DOOR))
          bit_true(sys_rt_exec_state, EXEC_SAFETY_DOOR);
      #endif
    #endif
```

and this would be added to cpu_map.h somewhere around line 91 or so:
```
  // Enables support for having a separate safety door input.
  // Disabled by default because the standard Arduino Uno pin mapping does not have the extra pin required.
  // #define SEPARATE_CONTROL_SAFETY_DOOR_PIN
```

chamnit, 2019-08-10:
On the 328p version of Grbl, yes, the safety door pin replaces the feed hold pin due to lack of available pins. It was intentionally written this way for a vendor that requested this option.

chamnit, 2019-08-10:
I can take a look at cleaning this up a bit to avoid future confusion but config.h does state that the feed hold pin behaves as a safety door switch. I never intended for user to have separate feed hold and safety door input pins on this build. 

With the Mega branch and if I recall correctly, the safety door pin and feed hold pin are on separate pins. So this issue is moot there. 

As far as updating the Mega branch, yes it’s a little behind. I intend to update everything when this dual axis feature is vetted and released.

bdurbrow, 2019-08-10:
Umm... OK...

> So this issue is moot there.
Well, it would be if ENABLE_SAFETY_DOOR_INPUT_PIN didn't exist on grbl-mega... so... I figure lets do that?

I'll completely delete it from my fork, and to make it easier for you, I'll generate a new fork and pull request for grbl-mega that fixes both the ISR and the config files (my existing fork has wandered so far off of the official grbl-mega that I don't think trying to make a new branch is going to be easy there... thus the new fork just for making patches that are intended to go back to the main official repository).

On '328, at the moment I'm thinking that it would be best if CONTROL_SAFETY_DOOR_BIT was just deleted, and in the main body of the code wherever CONTROL_SAFETY_DOOR_BIT did appear; to replace it with CONTROL_FEED_HOLD_BIT... this would remove CONTROL_SAFETY_DOOR_BIT as a tempting target for re-pinning.

Does that seem reasonable?

chamnit, 2019-08-10:
In the Mega branch, you're right that ENABLE_SAFETY_DOOR_INPUT_PIN is present in the config.h, but it doesn't do anything. Already deleted for the next push to the Mega branch. Regardless, the code itself is correct and assumes separate pins for each, as I told you. So, again, moot point.

Feel free to do whatever you wish to your Grbl branch. It's all open source. And I'll keep doing what I feel is right for this repo.

ecncshopInd, 2020-02-08:
Looks like the FeedHold pin is having NO behaviour ( Normally open ) which is not safe for a SAFETY_DOOR_InterLOCK kind of scenario ( A loose contact or a rat bite of the cable will render the safety interlock useless) . What is the best way to give it a NC behaviour ? Will enabling the ENABLE_SAFETY_DOOR_INPUT_PIN in config.h achieve this ?

ecncshopInd, 2020-02-16:
Awaiting reply. Limit switch has options for configuring as NC. What about this Safety_Door_Interlock signal to FeedHold pin A1. How to give NC (Normally Closed) behaviour to this signal ? Is there any option in GRBL ? Any code change will achieve this ?