gnea/grbl #707: Realtime jogging question

AlexHolden, 2019-08-17:
Hi, I'm adding support for the 1.1 jogging mode to my sender, and there's a detail I'm not sure I understand correctly. Say I am sending a stream of lots of small jog commands in a loop, and the user lets go of the jog button so I want to immediately stop jogging. What if I have already transmitted part of the next jog line when the event comes in? Will grbl flush the serial buffer when it processes the jog cancel command? Or do I need to finish sending the rest of the $J= line and wait for it to be acknowledged with 'OK' before I can safely send the jog cancel command?

zeevy, 2019-08-17:
Check this thread it may be helpful to you
#95

AlexHolden, 2019-08-17:
> Check this thread it may be helpful to you
> #95

Thanks @zeevy, I hadn't spotted that one. Sounds like it is not a simple issue.

closed 2019-08-17 by AlexHolden