gnea/grbl #711: Homing vs Limit Switch

rynaldi95, 2019-08-28:
Hi everyone. My name is Rynaldi. I'm currently working on a laser engraving machine project. 
I am using LaserGRBL as the software. But, I actually have a question about the homing and hard limit switch in GRBL.

I have installed NC limit switches in my machine in series connection as explained in the Wiki page of GRBL and the hard limit is working fine. I mean, when the limit switch is triggered the system stops. Actually, I thought that I can use these switches as the homing switches as well. Since I currently still have a 2 axis configuration, I modified the library a bit to be able to the X-Y homing sequence. When I trigger the homing with the $H command, the homing process was working fine until the end. The machine pulled back a bit as defined in the pull-off distance configuration. 

However, when I give the G0 X0 Y0 command, both axis will always hit the limit switch again and hence, trigger the hard-limit instead of going to X0 Y0 position.

Is there any solution how to solve this problem?

Paciente8159, 2019-08-29:
Hi there,

I think a G10 command is your awnser.
Check [this](https://github.com/grbl/grbl/issues/613#issuecomment-76854859).

biasedlogic, 2019-10-15:
@rynaldi95 this is the expected behaviour, even if I too find it desirable to at least optionally set machine 0,0,0 at the pull-off position and not at the switch position.
Work coordinates and offsets are anyway the way to work in CNC space, as machine 000 is only an arbitrary (albeit with a homing routine repeatable) reference frame that has little to do with where your workpiece is residing anyway.