gnea/grbl #715: Why is GRBL a library?

adammhaile, 2019-09-03:
Serious question that I can't seem to find the answer too...
Having GRBL packaged as a library makes it hard to have multiple versions of it on my system. I have to move directories around to be able to build a different fork of it. But I played around with it last night and was able to just create `grbl.ino` in the `grbl` directory with just `#include "grbl.h"` and now it works fine as a normal Arduino project. 
So was there a specific reason it was decided to package it as a library instead?

chamnit, 2019-09-03:
Don’t remember exactly. But if it can be compiled like you say, then it does simplify things for the end user quite a bit.

adammhaile, 2019-09-03:
@chamnit - yeah, it works great with only that one modification. I've got a bunch of different machines I'm working on and some of them have a fork of grbl that supports servo control instead of the Z-axis which is why I needed a couple different versions at the same time.
I'm also trying to setup a Travis CI pipeline that can automatically build firmware binaries for a bunch of different versions of `config.h` (and potentially the various forks I mentioned), which was easier having it as a normal project and not a library.
If any of that would be of any interest to you I'll report back when I'm done.

bdurbrow, 2019-09-04:
@chamnit -- I have a vague recollection that it had something to do with the build process on an earlier version of the Arduino IDE? Maybe?

@adammhaile  -- I would find anything you figure out interesting. I also have a bunch of very different machines; and putting together some sort of build manager script for dealing with all the different config.h & cpu_map.h variants was on my to-do list.

adammhaile, 2019-09-04:
@bdurbrow - ahh, yeah... the older Arduino versions were weird with large code bases. That may have been it.
For the basic user, making it no longer a library is super easy. I can submit a PR with that change shortly.
For the build manager stuff, I'm still working on it. I've got multiple config.h files working though didn't think about cpu_map.h - honestly never looked at that file. I can take a look. But I think doing multiple config.h files is top on my priority list. 
If ya'll are ok with leaving this issue open for now we can use it as the central point of communication on this.

drf5n, 2023-02-25:
> But I played around with it last night and was able to just create `grbl.ino` in the `grbl` directory with just `#include "grbl.h"` and now it works fine as a normal Arduino project.

It doesn't even need that much.  A simple empty `grbl.ino` file works.  

This make a working setup for a compilation:

```
mkdir GRBLTEST
cd GRBLTEST
git clone git@github.com:gnea/grbl.git
touch grbl/grbl/grbl.ino
```