gnea/grbl #718: How would it be possible to read PRB values in GCode after probing

BernardG, 2019-09-06:
When I probe with my macro, I see that GRBL answers with PRB values:

Is there a possibility to read theses (from what I can see, they are absolute positions at the probing point detected) in G-Code?

To be clearer (I hope), if I want to use G10 L2 Px instead of G92 after probing, I need to pass (if I understand properly) MACHINE positions, instead of current positions. How would I do that?

Thanks for your help.

MeJasonT, 2019-09-06:
$#

from :
Other Messages:

[PRB:0.000,0.000,1.492:1]: This is a little out of place, but as a service to GUIs, Grbl will immediately send a feedback message containing the triggered probe position upon a successful G38.2 straight probe command. This data can also be viewed in the parameters print out called by $#.

found here
[https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl](url)

produces the following output

Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
[G54:0.000,0.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
ok

to write a move to machine coordinates use
G53 X*** Y**** Z**** where * = coordinates required for move in negative space.

you will need to parse output to get the PRB info on its own

chamnit, 2019-09-06:
Please try to avoid posting in multiple places. There is a duplicate on the old site that has already been answered.

closed 2019-09-06 by chamnit

BernardG, 2019-09-06:
Sorry,did not realize my first post was on the old site!