gnea/grbl #720: Tool GCode commands

MeJasonT, 2019-09-11:
I'm sorry if this has been asked before , i find gits search engine crap - so didn't find anything.
When a tool change command is sent ie T106
is it possible to get grbl to respond with ok instead of error
its highly likely that all postprocessors will be written to send tool changes and grbl must be able to respond better than it does.
If It simply did nothing but sent out ok then the operator GUI software can intercept the commands.  The memory usage of simply saying ok should not be an issue
where could i add the acknowledgement

MeJasonT, 2019-09-11:
Candle handles the tool change but Grbl_panel does not.  I found in the reports..h .c that the tool change is reported on ?

langwadt, 2019-09-11:
if the GUI software can intercept the commands, it can also skip sending them ..

closed 2019-09-16 by MeJasonT