gnea/grbl #721: Considered switching to platform.io?

blindMoe, 2019-09-11:
Has any consideration gone into converting the project to be able to be compiled with platform.io?  It would be nice to have the ability to compile to more processor types.  Marlin has done a good job of implementing this in Marlin 2.0 and it is what I currently run on my CNC machine.  I would love to switch over to Grbl but I would like to retain the use of my SKR Pro 1.1 board since a) I already have it and b) I like the design of the board.

mike-pittelko, 2019-10-07:
Tried SuperGerbil?  Already ported to ST32.

hdo, 2019-10-07:
> Has any consideration gone into converting the project to be able to be compiled with platform.io? 

Please not. I would rather prefer simple old Makefile. I tried plattform.io and it's a behemoth. Just my opinion.

RealTadango, 2019-12-02:
You can add a few files and compile using platform.io. Simply add a src dir with main.cpp, pio config file and include grbl.h in main.cpp. Works for me :)