gnea/grbl #724: Camtool CNC v3.3 - few questions

zygmunt, 2019-09-13:
Hi,

I have Camtool CNC v.33. board, and here are my questions:
1. Do I have to fix cpu mapping in header file to get grbl updated on my board, or default one are correct?
2. Anyone has some random reboot when board is working? From time to time my board is rebooted/reseted. I don't know why. I know that is reset becasue linux kernel logs shows me usb reenumeration.
3. I didn't found any list of compatible boards in wiki. Would be good to have it (in case that I want to put to trash camtool v3.3) I understand that every board with atmega will works. But would be good to have list with tested devices.
4. Is there mailing list for grbl project, ot "Issues" in github is fine for every question? 

*) In case of questions for my settings:
`$$ < $0=10
$1=25
$2=0
$3=5
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=1000.000
$111=1000.000
$112=1000.000
$120=30.000
$121=30.000
$122=30.000
$130=200.000
$131=200.000
$132=200.000
ok`

zygmunt, 2019-09-13:
ad 1) Looks like not
ad 2) looks like version v1.1h.20190825 fixed problem
overall would be good to know, and get answer from you guys

Tahutipai, 2019-09-23:
+1 for question (4).  Is there a mailing list?

ulab, 2019-10-22:
> ad 1) Looks like not

Is that the answer to the first part 

> Do I have to fix cpu mapping in header file to get grbl updated on my board, 

or the second part

> or default one are correct?