gnea/grbl #725: Tool height, offsets and work coordinates

MeJasonT, 2019-09-16:
I'm looking for tuition from some of you knowledgeable cnc operators out there.
what im trying to understand is how work height and tool heights are set and if both can co exist using the machine and work coordinate systems.
Is G38 on the MCO or WCO and when the results are saved where are they saved.
i dont really understand how a job started in G54 uses the saved probe locations, i cant see how they are assigned to G54 or G55 etc
if i set the work piece using probing then replace the probe with an end mill, then the probed zero Z is what a WCO location.  If i then do a tool height probe, as far as i can see it exactly the same as the probing for WCO.  If The tool height is probed in in the machine coordinates would the machine quite happily move to the job and the tool would be at work zero?

wlaw, 2019-09-16:
Jason,
not sure if this answers your question completely, but its all I have.

in the high end industrial systems and even some very crafty people we
could use a tool changer. The tool changer always allows the tool height to
be set at the same height.
So to say it this way, tool 1 will always be tool 1 and have the same
height. The same with tool 2 and tool 3 etc.......

with my Shapeoko II, I agave to manually change the tool and this gets
quite aggravating when you want to change tools and run a different cut.

The way I fixed this is to run a automated process to set the tool height
to the table.
In this way i would be able to change the tool, run the macro and then
start cutting again.
I made a small video of this simple system that uses the probe feature.
Only i make the switch activate and calculate the offset of the switch
activation to the top of my waste board.

https://www.youtube.com/watch?v=dRl4OHCeuMo

All the instructions were on this also.

Bill

MeJasonT, 2019-09-16:
Hi Bill,
Thank you very much this is very helpful.
My question was asking about the background of the procedure with the intention of then moving on to what you have so brilliantly described.
I just happen to be working on modifications the Grbl_panel so that it can parse out tool changes and then perform tasks to set up the new tool and return to the work-piece.
So far I can detect the T and M6 statements, grbl currently pauses when it discovers a tool change.  The machine moves using machine coordinate to the tool change position front left of the Table/work area so that a manual tool change can be carried out when a resume is issued the machine moves back to the work coordinates and continues with the job.
triggering a tool height check was where i had the issue, hopefully i will advance my mods to the GUI to perform the height.  I cant see any point at this stage of attempting to go fully automatic especially seen as i have not even started on a tool carousel/slotted holder and a change from the MT3 tool holder.

MeJasonT, 2019-09-16:
Bill can you pm me
cheers 
Jason

neilferreri, 2019-09-16:
I use CNCjs with some macros to do this. GRBL supports G43.1 dynamic tool length offsets, but I prefer just changing the WCO. That way, position for the current tool is retained across power cycles.
Here are my macros.
https://github.com/cncjs/CNCjs-Macros/tree/master/Initial%20%26%20New%20Tool

MeJasonT, 2019-11-03:
Hi Neil
is there any chance you could provide a list of the Gcode commands sent from CNCjs to grbl
during the tool change operation to help me get my head around how your macro works.
I'm not familiar with java or python but would like to write a tool change function to port into grbl panel.
Bill (wlaw) - As a user and frequent supporter of grbl panel i was hoping to make your acquaintance and ask for your assistance in making changes to Gerrits grbl panel.  some changes have already started/been implemented.  Auto Probing and multi-axis machine for both Lathe and 5 axis.
my coding is quite excusable for an abortion and gives Gerrit many sleepless nights i'm sure.

neilferreri, 2019-11-03:
@MeJasonT  I commented the macros, so you should be able to follow there.  Basically, after Z zero is set, you probe with the first tool.  Let's say your probe is a work Z21.65, CNCjs has the (unique?) ability to store that variable globally.  When you change tools, and the probe is done, you just set that same location to Z21.65 using a G10L20. Any subsequent tool does the same thing.
You can do the same thing without the global variables by probing current tool and new tool within the same macro.  This does the same thing, but you need to probe, change tools, probe on each tool change.
Here's the macro I used before the global variables were introduced.
https://github.com/cncjs/CNCjs-Macros/blob/master/Precision_ToolChange_Macro.txt

MeJasonT, 2019-11-03:
Thanks Neil,
I have downloaded the master branch from your GIT.
My plan is to parse the NC file into grbl panels code window and intercept T* tool change instructions.
Then using a dll in vb.net withing grbl panel to handle the variables and run a number of sequential macros to achieve a tool change.  Part of the code would be to record a number of pre set tools including holders in a tool table with associated tool length offsets.  initially the user will still need to manually change the tool instructed from the sub program/form but it would be nice to eventually get to a point where 4 or 6 tools could be in a slotted tool frame at tool change position (right hand side of the bed) and using a power draw bar to automate the tool change.
I will read through the link you have so kindly sent and try and get my head around it.
cheers
J.

neilferreri, 2019-11-04:
@MeJasonT 
You might be interested in these:
https://github.com/scottlaird/cncatc
https://github.com/ahedderich/cncjs/commit/49566ee6f87f6e417505c2b7502e5b766ccb7b0f

My method doesn't need tool tables or even pre-defined offsets.  It's not much different than just probing for Z after changing a tool...just "automates" it, and removes the variable of milling away your original Z0.

sjbaird, 2024-06-30:
So I'm sure that this will help. 
G10 L2 P1 Z0 sets the WCO to zero
G10 L20 P1 Z0 sets the tool offset relative to the WCO.
So set you WCO then probe your tool at your tool probe location. Run that tools code. Change tools and reprobe tool at tool probe location.