gnea/grbl #726: Laser burning deep on edges.

BainViljoen, 2019-09-17:
Hi,

I was previously using an Eleksmaker 2.5w laser and it was working perfectly.
I bought a 15 diode laser and i don't know if I'm doing something wrong with the config but it burns deep on the sides of any object i try to laser. 
For example if i try to engrave a word everything would be perfect except the outside edges of the first an last letter of the word would be burnt very deep. but the burn is kind of a slope starting about 1-3 mm from the side then being extremely deep on the edge.

Any advise would be greatly appreciated.
Please see below links for the 2 different lasers.

Eleksmaker: 
https://www.banggood.com/EleksMaker-EL01-2500-445nm-2500mW-Blue-Laser-Module-PWM-Modulation-2_54-3P-DIY-Engraving-Machine-p-1179592.html?rmmds=myorder&cur_warehouse=CN

Other Laser:
https://www.aliexpress.com/item/32994386595.html

Attached is the grbl config I'm using.
[grbl config.zip](https://github.com/gnea/grbl/files/3619346/grbl.config.zip)

tklus, 2019-09-17:
What software are you using to generate the gcode? Sounds like at the transition from left to right there is a slight pause.

I use laser etch from Jtech photonics, this software has an option for the machine to travel past the edge of the graphic that prevents over burning.

BainViljoen, 2019-09-17:
Mmmm sounds like the "hack" i ended up doing.... i added a 1px line on pothe edges and the image was left in a perfect state lol. Thanks for the advice though! I will check it out. Im using lasergrbl.

chamnit, 2019-09-18:
Make sure you have laser mode enabled in settings. Also, try using M4, rather than M3, to enable dynamic laser adjustments with axis speed. Should help with overturn into corners.

BainViljoen, 2019-09-18:
Setting the M3 to M4 was the trick!!! an now my engravings have more definition! thanks allot!!

closed 2019-09-18 by BainViljoen

TEKA1985, 2020-09-19:
> Hi,
>
> I was previously using an Eleksmaker 2.5w laser and it was working perfectly.
> I bought a 15 diode laser and i don't know if I'm doing something wrong with the config but it burns deep on the [...]

yes i got some problem like this.. in the end om edge the line burn so highly, whats the trick for this kind