gnea/grbl #728: Dual Lasers

BainViljoen, 2019-09-19:
Hi,

I have a suggestion, might be stupid...
Is it possible to add another laser and another y axis stepper.
Then they can split the commands, with the correct safeties so that they don't crash, after the line is processed it goes to the next line, on the x axis. both lasers power should be controlled independently.

BainViljoen, 2019-09-19:
Maybe there can be a limit switch on the one head and they work towards each other until the limit switch triggers then the one backs of out of the way and all commands for the working laser gets processed then all commands for the "clash" gets processed then they continue together on the next line?

mike-pittelko, 2019-10-06:
Would also require a second PWM, in addition to a second Y axis, which arduino doesn't have the pins for in the standard configurations.

beckmx, 2019-11-22:
@BainViljoen some chinese boards have two ports for two lasers https://es.aliexpress.com/item/4000185249095.html?spm=a2g0o.productlist.0.0.9ccc7bdcvvF2Ry&algo_pvid=cd8efc0a-eb78-4e79-8e8c-b0750a6bd808&algo_expid=cd8efc0a-eb78-4e79-8e8c-b0750a6bd808-2&btsid=092cc190-a805-48e7-909c-dc781693d97e&ws_ab_test=searchweb0_0,searchweb201602_1,searchweb201603_52

bdurbrow, 2019-11-25:
You might be able to do this on the Mega version; or one of the ARM ports, but the '328 processor is maxed out as it is.

However, it would be a major undertaking to do so - you would have to modify the g-code interpreter, motion planner, and step driver, as well as come up with some sort of post-processor to integrate the two g-code streams so that they moved synchronously.

It would be much easier to increase your cutting speed by putting that money on a more powerful laser module.