gnea/grbl #729: Homing Limit Switches XY axis aren't recognized when clicked.

Silvuurleaf, 2019-09-19:
Problem: I want to set up a homing cycle for my CNC machine and to do so I placed pairs of limit switches at both ends of all my axes, X, Y, and Z. Then I wired them in the suggested NC configuration. When I try to run the homing procedure several things happen.
1. The CNC machine moves in the positive direction that varies based on the pull of distance specified by $27 towards the upper right corner of my workspace. 
2. It stops and then goes again a certain distance.
3. Most of the time it completely ignores my limit switches and runs through them. Sometimes however it throws an ALARM 8 message saying I should change the pull off distance or check the wiring though.

Limit Switches are wired in NC setup mentioned here: https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

I have modified the config.h to only home the XY and Y-axis so I'm not doing Z at all. (I'm essentially pretending I don't have one right now, but I did wire them up in the circuit)

I have checked my limit switches with a multimeter and with the Arduino IDE setting $10 = 19 I can see which limit switches are being pressed. PN: XYZ. When I manually click any of the switches it correspondingly shows that they are being pressed in the IDE and the verbose statements in GRBL GUI.

These are my current settings in GRBL.

[image]

Side information
Whenever I try to enable hard limits it throws an Alarm 1 error saying they have been pressed even when they are not. Not sure if this is related.

if I change $5 = 0 then an Alarm 8 error goes off saying suggesting I check the wiring and increase the pull off distance.

Silvuurleaf, 2019-09-19:
Further testing it seems whenever I call $H it defaults into the pull-off mode as if the CNC has already pressed the switches. It repeats the pull-off movement twice then exits.

chamnit, 2019-09-20:
Do the status reports detect and show the correct limit switches when you trigger them manually?

beckmx, 2019-11-22:
@Silvuurleaf also increase the time of the pull-off, in the settings, and make sure you have capacitors in your limit switches, I used to have plenty of those problems, and as a workaround to these issues, I ONLY enable Homing BUT disable hard limits, this "ignores" the limit reached but only work on the homing cycle