gnea/grbl #730: Increase Spindle Dwell Time

adammhaile, 2019-09-20:
Without actually inserting a dwell command into the gcode is there any way to in firmware increase the amount of time it waits after an `M3 Sxxxx` command?
I'm using firmware modified to control a servo with M3/M5 commands and it mostly works but it barely hits the paper when it starts moving again.

neilferreri, 2019-09-20:
What modified firmware?  Doesn't the S value set the rotation angle of the servo?

adammhaile, 2019-09-20:
I cobbled it together from a couple. Cprezzi and some others.
You are right but I want there to be more time between hitting the paper
and starting toove again. It's missing bits of lines

neilferreri, 2019-09-20:
How are you generating the gcode? I'd think that's where the G4 would come from. Grbl just interprets it.

adammhaile, 2019-09-20:
I'm using LightBurn which is meant for lasers and it doesn't include G4 or
have any way to include it.
Really what I'm trying to do here is figure out how to set this up without
having to do any special gcode. Maybe it's not possible....

phil-barrett, 2019-09-22:
No prost processor with Lightburn? You could use a stream editor like SED to insert your G4s after the S commands.  It's an extra step but at least you don't have to physically edit the g-code file.

adammhaile, 2019-09-22:
It's on the long term desired feature list for LightBurn but would be a
long way off. And as noted earlier, I'm mainly just curious in finding if
it's possible to "solve" the problem in firmware.
I've looked through the code but can't really figure out how it determines
the spindle start up delay.

mike-pittelko, 2019-10-06:
It's certainly possible to solve in firmware, and is in the SuperGerbil version of grbl 1.1 as a $ setting for spindle spin up delay.  I think the default is already 1+ seconds, so sort of depends on what you're doing.  If I remember, lightburn supports "overscan" which can be used to help this too.

adammhaile, 2019-10-07:
Do you know where in regular grbl I could add this?

mike-pittelko, 2019-10-07:
not offhand, but I'd suggest looking at this to see what they did: https://github.com/paulusjacobus/SuperGerbil