gnea/grbl #732: Determine absolute position of second end stop

marshaul, 2019-09-27:
We can have two end stops for each axis -- one for each end -- and presumably use these for hard limits.

What if we want to use the second switch to determine the position of the far end relative to the home end? That is, what if I want to know the "absolute" position of the second end stop?

What is the best way to achieve this? Modifying GRBL is acceptable in this application. #

MeJasonT, 2019-09-27:
You could home the normal negative coordinate work space as normal - save that as a work coordinated ie G59
then change the grbl setting for homing direction then re home the machine.
you should then have two sets of coordinates for home switches at each end of the work area.
Can you explain why you are asking to do something that isn't normal practice.

marshaul, 2019-09-27:
I haven't tried saving it in i.e. G59 but I don't think this will work. This was my original intent, but when I queried the position it seems that the coordinates are always re-zeroed so that, when homing towards 0, the Z value becomes 0 - the homing pull off, and when homing in the other direction it becomes -(Z max travel - homing pull off). 

I think I probably need to use the probe functionality for this.

This is a non-standard application (it's built on a CNC engraver but it isn't engraving or milling), and I'm forced to retrofit a solution into existing equipment, and learn all this as I do it.

MeJasonT, 2019-09-27:
are you asking to home on one set of end stops or the other or are you asking to know the position of both.  When you run a job grbl will only work from one set of homed coordinates and will count either incremental or absolute moves from that homed position.

marshaul, 2019-09-27:
I want to know the position of one end stop relative to the position of the first end stop (the home coordinate).

MeJasonT, 2019-09-27:
ok chap
so you want to know where your zero is from your home end stop
ie setting $27 pull off distance ?

MeJasonT, 2019-09-27:
why

marshaul, 2019-09-27:
No, the zero is at the home end stop -$27.

I want to know the exact distance between end stops.

But I think the best way to do this is re-wire the second one as a probe.

MeJasonT, 2019-09-27:
I'm curious and wanting to help you find a solution.  
Normal practice is to work from a known position, grbl cant handle knowing where it is between two so it works everything from the one home location (it knows where this is).  If you are wanting to measure your bed then there is a simpler method.  your work area once calculated will remain constant and it is highly unlikely your second end stop will magically move.
Do you have your motor steps setup and calibrated- ie you move 10mm it measures 10mm

marshaul, 2019-09-27:
The solution is to short A5 to D11/D12 (depending on your CNC shield), in my case I jumpered the Z+ header to SCL, as labeled on my CNC shield, and then run G38.2 Z-whatever

MeJasonT, 2019-09-27:
is that not just simply zeroing the axis at the 2nd end stop position
i'm sure you are aware that you can tell grbl where your 2nd limits are by entering the max travel settings $130,$131,$132 if you use soft limits then that's the end of the world, if you use hard limits they will be set to trigger at that location ?

marshaul, 2019-09-27:
Not sure what you are trying to say about the behavior of $130 etc. when using hard limits instead of soft.

chamnit, 2019-09-27:
You can temporarily connect the second limit switch to the probe pin A5. Use the probe cycle to locate the exact trigger position, that is after you’ve homed the machine to the other limit switches.

marshaul, 2019-09-27:
Why do you say "temporarily"? Both functions seem to work with this configuration. After all, the internal pullups are pretty weak.

MeJasonT, 2019-09-27:
Sonny superstar woo hoo

chamnit, 2019-09-27:
Perhaps because you’d like to use the probe pin for its intended purpose, to probe tool offsets to simplify doing jobs that require multiple tools. And once you know exact distance, it is likely to not change once set. Otherwise, no particular reason if you find it working out for you.

MeJasonT, 2019-09-27:
Once you know this location what use is it other than recording the max travel ?

Planned purpose for probing of electronics pcb using bed of nails and using grbl as a 3 axis robot to automate testing.
are pogo test pins all at the same height
like Sonny says once the location is probed grbl will happily probe to that limit provided nothing changes.

marshaul, 2019-09-27:
The pogos are all at the same height. Unfortunately, the guys who designed the spindle/tool holder adapter for the PCB didn't take into account absolute positioning. And I know these guys, they will want to disconnect everything and look at it any time there is a problem. So I decided we need a way to automatically determine the tool position. It's really a probe, but it's also conceptually something of an end stop in that the tool should never exceed that position, whatever it is. 

Anyway, thanks for the responses. Talking it through always helps.

MeJasonT, 2019-09-27:
I guess its the z axis we are discussing
if you set the Z axis Zero height manually 9not using a probe and come down onto the pogo pin.
set the contact as zero
and then set a software limit 0.5mm below the zero (if the pogo will allow 0.5mm of compression)
then if the z accidentally exceeds zero then it should trigger the soft end stop and cancel any operation.
using the gerber file it should be possible to get exact x and y coordinates and convert those to gcode. there are a few guys who have used grbl for pcb drilling and milling so it shouldnt be rocket science to find a a method of working between vias as if your were drilling a pcb but probing instead.
This should be possible using a standard grbl setup and automating the contact data from the test probe.  Ie. the meter probe makes contact with the pcb - using a seperate arduino or whatever to send the probe value to excell and then code excell to record/format the values in the record format you are looking for.  the pattern will be repeatable, so when the board test is finished putting a second board in a jig which locates it at work zero location the code will go through the same cycle again.