gnea/grbl #733: Laser On-Off issue.

Not-invented, 2019-09-29:
Not sure if it's the correct place to ask since i don't see the library I'm going to ask about here, but hope someone will answer. 
I have CNC 3018 machine and using it's laser cutter with Fusion 360. I downloaded the lib called "grbl Laser.cps"(there is a link to this page inside the library) from the link Fusion 360 provided. The problem is that the code which is generated by this lib doesn't include laser power on-off commands. So when I want to cut more than one countour I get additional cuts on the laser's way between them. Simulation in Fusion looks just fine and the files generateted do have S255 and S0 commands at the beginning and at the end of file only.

vol4enok, 2019-10-03:
This script is designed for firmware grbl 1.1 version. There is a laser mode. It is included by the command $32=1. In this mode, grbl turns off the laser when executing displacement commands (G 0). Therefore, commands to turn the laser on and off are not needed. More https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#32---laser-mode-boolean

Not-invented, 2019-10-05:
> This script is designed for firmware grbl 1.1 version. There is a laser mode. It is included by the command $32=1. In this mode, grbl turns off the laser when executing displacement commands (G 0). [...]

Thanks for clarifying. That explains it. But I have rewritten the library already, so it turns the laser off and on when needed:D

mike-pittelko, 2019-10-06:
The difference is M4 vs M3.  With laser mode enabled, M4 controls the laser on and off with movement.  M3 does not.  But you also need laser mode enabled for that to work.  If you rewrote the library, you may have broken this with M3.  With M3, the laser will stay turned on at the same power level during acceleration, which you probably don't want (uneven burn), and will stay on at corners (heavy burn in a corner).

Samlokihel, 2020-09-20:
boa noite, poderiam me ajudar com o erro 17? sofrendo muito com isto......