gnea/grbl #735: Question: Using laser with 3.3V PWM

cnburke, 2019-10-02:
I've been digging through the wiki and source, but I'm still not clear if I'd be able to use a laser with a 3.3V PWM. The docs say that max output is 5V, but is there an option to set the cap at 3.3V? I'm sure that I'll loose resolution, but that's ok.

PicEngraver, 2019-10-03:
I believe an answer to your question lies entirely with the laser driver you are using.  Lasers are current-driven devices, not voltage.  So if the max input to your driver is designed, or adjustable, to be a 3.3 volt PWM signal to get the max current out to the laser, then all should be good.

cnburke, 2019-10-03:
Yes, the driver is designed to accept a 3.3V 20kHz PWM signal. My only concern was with the 5V output of the spindle/laser PWM pin. I may have to use a level shifter to get it to work, but I think it will.

bdurbrow, 2019-10-05:
You need a level shifter; but it can be as simple as a voltage divider made from a couple of resistors.

cnburke, 2019-10-14:
Thanks! I went ahead with the proposed level shifter that @bdurbrow suggested, and that looks great! 

The only point I've been trying to figure out now is the PWM frequency. I've tried changing the pre-scalar in the cpu_map.h file, but I haven't seen any results. Any suggestion? I'm using a scope to verify the output of the signal.

closed 2019-10-14 by cnburke