gnea/grbl #737: Command for R support

beckmx, 2019-10-09:
Just wondering, I searched in the shapeoko forums and some say yes but not quite sure if the R command is supported by grbl. I have a piece of gcode with this command and before I send it I just want to make sure it might probably work:

```
G90
G0 X0.0000 Y0.0000 Z0.8000
(Dragknife Toolpath3)
T1
(Dragknife End Mill)
F50.0
F20.0
S0
G0 X0.6291 Y1.4893 Z0.2000
G1   Z-0.0500 F20.0
G3 X0.3364 Y2.1962 R1.0000 F50.0
G1   Z-0.1400 
G1 X7.4141 Y9.2703 
G0   Z0.2000
G0 X9.7718 Y9.7373 
G1   Z-0.0500 F20.0
G3 X8.3572 Y8.3234 R1.0000 F50.0
G1   Z-0.1400 
G1 X1.2794 Y1.2494 
G0   Z0.2000
G0 X6.0004 Y9.2749 
G1   Z-0.0500 F20.0
G3 X7.4141 Y9.2740 R1.0000 F50.0
G1   Z-0.1400 
G1 X10.0048 Y6.6781 
G1   Z-0.0500
G2 X8.5911 Y6.6790 R1.0000 
G1   Z-0.1400 
G1 X2.2207 Y0.3119 
G0   Z0.2000
G0 X-1.3138 Y1.5812 
G1   Z-0.0500 F20.0
G3 X0.0983 Y1.5787 R1.0000 F50.0
G1   Z-0.1400 
G1 X0.6582 Y1.0150 
G0   Z0.2000
G0 X3.0168 Y-1.3446 
G1   Z-0.0500 F20.0
G2 X1.6050 Y0.0720 R1.0000 F50.0
G1   Z-0.1400 
G1 X1.0469 Y0.6300 
G1   Z-0.0500
G2 X2.7540 Y-0.0771 R1.0000 
G0   Z0.2000
G0 Z0.8000
G0 X0.0000 Y0.0000

```

barma1ey, 2019-10-09:
yes, grbl does support radius format arcs. But it seems your code won't run because arcs constructed incorrectly

chamnit, 2019-10-09:
Yes. And don’t use R mode arcs. They are way more sensitive to numerical errors.

closed 2019-10-09 by chamnit

beckmx, 2019-10-09:
should I replace it with G91.1? that's the default output of aspire so I am not sure what should it be

barma1ey, 2019-10-09:
the syntax is correct,  but values contain numerical errors.  and G91.1 command is not supported

beckmx, 2019-10-09:
For instance this line (the first appearance of the -R-)
`G3 X0.3364 Y2.1962 R1.0000 F50.0`
@barma1ey what are the numerical values? the decimals? (sorry for the questions)

beckmx, 2019-10-09:
@chamnit I am starting to use Aspire (https://www.vectric.com/products/aspire) but if you point me into an example I will dig in there to make the conversion

MeJasonT, 2019-10-09:
Are you using a postprocessor written for GRBL, it should be outputting your file in the correct format for GRBL.

beckmx, 2019-10-09:
@MeJasonT yeah you were right, the postprocessor option was the failure :s, now exported correctly

MeJasonT, 2019-10-09:
You are very welcome