gnea/grbl #738: QUESTION: Modifying Laser/Spindle Frequency to ~20-kHz

cnburke, 2019-10-14:
I hate to open another issue for a question, but here goes:

Looking at trying to modify the PWM output of the variable spindle PWM to get close to 20kHz. The pre-scalars don't quite offer up what I'm hoping to achieve. I'm trying to understand the results of meddling with the PWM timer registers, but I'm also worried I'm going to affect something negatively, as all of the timer resources are tightly managed. 

Is there any direction that could be given towards being able to modify the PWM frequency to a more custom frequency?

beckmx, 2019-11-22:
I would put another atmega (tiny at least) in the port to avoid messing with the main code, that would allow you to update its firmware in a controlled way

bdurbrow, 2019-11-25:
Can you switch to an Arduino Mega (and the Mega version of Grbl)? The timers on the ATMega2560 are more flexible...

rafaelalvesferreira, 2020-01-20:
@cnburke, I'm on the same page. I'm looking for a laser module that the PWM operates in 3.3V (problem solved with a Bi-Directional Logic Level Converter) and a frequency between 20-50kHz. 

Without the prescaler we can get 62.5kHz on the fast PWM mode. Looking better at the Avr datasheet I found that using the phase correct PWM mode we get half of it (31.25kHz) because the counter goes from 0 to 255 and back to 0. This way I'd be within the working range. 

What I wonder, and it's my doubt, changing from fast PWM to phase correct PWM have any effect on the overall code? The fast PWM was choosen by any specific reason?

Tks

jak-pan, 2020-02-04:
I'm also looking for 31.25kHz because my controller needs 20+kHz to operate.