gnea/grbl #739: What happens if the sum of maximum speeds needs more than 115200 Hz?

MARIOBASZ, 2019-10-14:
I have this doubt: what happens if the sum of maximum speeds needs more than 115200 Hz?
suppose that for G0 it gives me 600RPM and I use 1/32 micro steps, it is equivalent to 64000 pps or 64khz.
If I want to simultaneously move two axes at maximum speed, what is the problem?
Synchronization problem / positioning error?
or
Will they run slowly but there will be no posición problem?
Should I modify baudrate?
Is my reasoning correct?

biasedlogic, 2019-10-15:
You are mixing up two things.
Baudrate has nothing to do with maximum speed of motor pulsing. 115200 is
the brutto bitrate of communication link
Maximum pulse rate is about 30kHz at any moment, per axis (it doesn't
matter if only one moves or all three move).
If you try exceeding it, the whole thing will crash, i.e. hang up, reset,
reinitialise.
You are required to calculate your maximum speeds upon configuration, it's
not checked live in operation
Marek

MARIOBASZ, 2019-10-15:
Thanks you. Regards

CNCmaisdeMILL, 2019-11-02:
i use esp 32 Which baudrate should I configure?
I would like 200khz to use high resolution servos.