gnea/grbl #743: Invalid machine position on 360 degrees arc motions

Overdrivr, 2019-10-21:
Hi,

We're using GRBL quite intensively and we recently discovered an issue with GRBL producing invalid motions for rare circular motions.

When an arc motion has the same start and end, bCNC interpretates this motion as a full 360 degrees arc, while it looks like GRBL randomly interpretates this to a 0 deg or 360 deg arc (maybe numerical stability issue ?). By randomly, I mean different rotations (0, 90, 180, 270) of the same part results in different GRBL motions.

For instance, the motion below should represent as final move a full 360 degrees but results in a zero degrees motion:

```
(...)
G01 X-1964.503 Y123.761 Z-18
G03 X-1965.409 Y127.282 Z-18 I-5.235 J0.530
( ARC MOTION BELOW HAS SAME START AND END)
G03 X-1965.409 Y127.282 Z-18 I-9.091 J-6.282
(...)
```

In bCNC, the motion in question results in a perfect circle:

[image]

But in GRBL (simulated with a python script that feeds commands and parses status), it results in a different motion (the last arc is missing) with a 90 deg rotations:

[image]

But with a 270 degrees rotation the motion is correct:

[image]

chamnit, 2019-10-22:
Thanks for the report. Likely numerical error as you say. Can you provide your Grbl settings, $I build info output, and any details of any changes you have made to the firmware from default? It's way easier to find the problem when I can reproduce it here.

Overdrivr, 2019-10-22:
Sure thing, this is happening with latest GRBL 1.1h, no modifications to the source code.

```
$0 | Step pulse, microseconds  | 10
$1 | Step idle delay, milliseconds  | 25
$2 | Step port invert, mask | 0
$3 | Direction port invert, mask | 6
$4 | Step enable invert, boolean | 0
$5 | Limit pins invert, boolean | 0
$6 | Probe pin invert, boolean | 0
$10 | Status report, mask | 3
$11 | Junction deviation, mm | 0.010
$12 | Arc tolerance, mm | 0.002
$13 | Report inches, boolean | 0
$20 | Soft limits, boolean | 0
$21 | Hard limits, boolean | 0
$22 | Homing cycle, boolean | 0
$23 | Homing dir invert, mask | 0
$24 | Homing feed, mm/min | 25
$25 | Homing seek, mm/min | 500
$26 | Homing debounce, milliseconds | 250
$27 | Homing pull-off, mm | 1.000
$30 | Max spindle speed, RPM | 1000
$31 | Min spindle speed, RPM | 0
$32 | Laser mode, boolean | 0
$100 | X steps/mm | 66.100
$101 | Y steps/mm | 116.300
$102 | Z steps/mm | 642.500
$110 | X Max rate, mm/min | 15000
$111 | Y Max rate, mm/min | 15000
$112 | Z Max rate, mm/min | 3000
$120 | X Acceleration, mm/sec^2 | 80
$121 | Y Acceleration, mm/sec^2 | 80
$122 | Z Acceleration, mm/sec^2 | 50
$130 | X Max travel, mm | 200
$131 | Y Max travel, mm | 200
$132 | Z Max travel, mm | 200
```

```
$I
[VER:1.1h.20190825:]
[OPT:V,15,128]
ok
```

And here is an NC file to reproduce:

[EP18_2500x1250_Batch2_Plaque3.main[d=7.9mm].nc.zip](https://github.com/gnea/grbl/files/3754247/EP18_2500x1250_Batch2_Plaque3.main.d.7.9mm.nc.zip)

Overdrivr, 2019-10-22:
Sorry, misclick

bdurbrow, 2019-10-24:
I'll look into it also when I get a chance. It at the moment, I'm sitting in a restaurant though with no Arduino on me...

Um... just as a workaround; can you edit your postprocessor to generate multi-segment arc moves for this case (so a full circle would be output as perhaps two or three moves)?

Overdrivr, 2020-01-15:
@bdurbrow That's what we ended up implementing. 

I can help fix this issue, but is there a test suite to write a regression test for this bug ?