gnea/grbl #744: GIT site related

MeJasonT, 2019-10-23:
Why cant i upload images using google chrome to this forum
keep getting
"something went really wrong, and we cant process that file"
which annoys me on three counts.

I cant post images to help people
It was only a 40k jpeg 
and whats worse they put a comma before the and in the error message

any clue as to why GIT is being a git.

winder, 2019-10-23:
Unfortunately the only thing you can do is try a different browser and ignore offensive grammar.

MeJasonT, 2019-10-23:
Cheers Winder
one is easier than the other the browser i can cope with.

MeJasonT, 2019-10-23:
I used DuckDuckGo it obviously works
thanks again winder

[image]

closed 2019-10-23 by MeJasonT