gnea/grbl #745: Continuous jogging in UGS

breiler, 2019-10-23:
I've been struggling trying to implement continuous jogging in UGS in such a way that if the user holds a jog button it will enter continuous mode and should stop once the button is released. 

So it seems to work most of the times, we queue jog commands filling the buffers one command at the time in a fixed interval as long as the jog button is pressed. Once the button is released we'll send a jog cancel command and it stops like expected. 

But I find it difficult to find a good balance keeping the planner buffer filled in GRBL so it will keep jogging at the desired feed rate, and not filling it. It seems like once it's filled it will not be able to cancel and completely clear the planner buffer. I tried following the discussion here #95, but there was no conclusive solution?

When I start jogging the commands are filling the buffers:
```
[verbose] <Idle|MPos:268.290,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:268.290,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:268.290,0.000,0.000|Bf:15,128|FS:0,0>
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:268.310,0.000,0.000|Bf:14,128|FS:84,0>
>>> $J=G20G91X1.667F1000
ok
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:268.620,0.000,0.000|Bf:12,128|FS:186,0|WCO:0.000,0.000,0.000>
>>> $J=G20G91X1.667F1000
ok
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:269.340,0.000,0.000|Bf:10,128|FS:300,0|Ov:100,100,100>
>>> $J=G20G91X1.667F1000
ok
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:270.480,0.000,0.000|Bf:8,128|FS:426,0>
>>> $J=G20G91X1.667F1000
ok
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:272.040,0.000,0.000|Bf:6,128|FS:546,0>
>>> $J=G20G91X1.667F1000
ok
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:274.020,0.000,0.000|Bf:4,128|FS:672,0>
>>> $J=G20G91X1.667F1000
ok
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:276.420,0.000,0.000|Bf:2,128|FS:792,0>
>>> $J=G20G91X1.667F1000
ok
>>> $J=G20G91X1.667F1000
ok
[verbose] <Jog|MPos:279.230,0.000,0.000|Bf:0,128|FS:912,0>
>>> $J=G20G91X1.667F1000
[verbose] <Jog|MPos:282.460,0.000,0.000|Bf:0,128|FS:1038,0>

**** Canceling file transfer. ****
```
Now the cancel jog command (0x85) was issued and it will decelerate:

```
[verbose] <Jog|MPos:286.060,0.000,0.000|Bf:0,128|FS:1110,0>
[verbose] <Jog|MPos:289.690,0.000,0.000|Bf:0,128|FS:990,0>
[verbose] <Jog|MPos:292.950,0.000,0.000|Bf:0,128|FS:864,0>
[verbose] <Jog|MPos:295.730,0.000,0.000|Bf:0,128|FS:744,0>
[verbose] <Jog|MPos:298.130,0.000,0.000|Bf:0,128|FS:624,0>
[verbose] <Jog|MPos:300.170,0.000,0.000|Bf:0,128|FS:498,0>
[verbose] <Jog|MPos:301.780,0.000,0.000|Bf:0,128|FS:372,0>
[verbose] <Jog|MPos:302.890,0.000,0.000|Bf:0,128|FS:258,0>
[verbose] <Jog|MPos:303.640,0.000,0.000|Bf:0,128|FS:132,0>
[verbose] <Jog|MPos:303.990,0.000,0.000|Bf:0,128|FS:0,0>
```

But then I'll receive an ok-message and start accelerating again and it will keep running:
```
ok
[verbose] <Jog|MPos:304.070,0.000,0.000|Bf:14,128|FS:96,0>
[verbose] <Jog|MPos:304.490,0.000,0.000|Bf:14,128|FS:216,0>
[verbose] <Jog|MPos:305.330,0.000,0.000|Bf:14,128|FS:336,0>
[verbose] <Jog|MPos:306.570,0.000,0.000|Bf:14,128|FS:456,0|Ov:100,100,100>
[verbose] <Jog|MPos:308.210,0.000,0.000|Bf:14,128|FS:576,0>
[verbose] <Jog|MPos:310.260,0.000,0.000|Bf:14,128|FS:696,0>
[verbose] <Jog|MPos:312.740,0.000,0.000|Bf:14,128|FS:822,0>
[verbose] <Jog|MPos:315.630,0.000,0.000|Bf:14,128|FS:942,0>
[verbose] <Jog|MPos:318.940,0.000,0.000|Bf:14,128|FS:1062,0>
[verbose] <Jog|MPos:322.670,0.000,0.000|Bf:14,128|FS:1188,0>
[verbose] <Jog|MPos:326.850,0.000,0.000|Bf:14,128|FS:1308,0>
[verbose] <Jog|MPos:331.350,0.000,0.000|Bf:14,128|FS:1428,0>
```

I know that in this example I'm using a too large step size and an unreasonable feed rate, but this was only to illustrate a problem that would otherwise be intermittent.

We could probably work around this issue in UGS and keep track of the planner buffer utilisation. But this would require our users to have the setting $10=3 so I don't find that optimal. Do you have any other ideas?

zeevy, 2019-10-23:
'''
We could probably work around this issue in UGS and keep track of the planner buffer utilisation. But this would require our users to have the setting $10=3 so I don't find that optimal. Do you have any other ideas?
'''
I think this is only the best solution currently which was implemented in android app "Grbl Controller" and working fine

breiler, 2019-10-23:
Yes it seems to work by parsing out the size of the planner buffer and make sure not to fill it.
It's not elegant, but if it is the intended behaviour or if can't be fixed in a better way then I guess that's ok.

In that case, may I write a couple of lines on the wiki about this?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

bdurbrow, 2019-10-24:
What I did for the "auto-feed" on my jogging mode is to not send short moves, but when the key is pressed to start moving, issue a jog command to move to the far extent of the axis' travel; and when you push the stop key it issues a jog-stop command.

In the specific case of UGS; does the API that you are working under give you the distinction between key-down, key-repeat and key-up events? If so, what I'd do is this:

On Key-Down: send a move of the current jog step size.
On Key-Repeat: if it's the first event since the Key-Down, send a move all the way to the far extent of the axis.
On Key-Up: send a jog-cancel command.

AlexHolden, 2019-10-24:
The way I'm doing it is on a button down (or joystick moved) event, I switch the sender thread into a special 'realtime jogging mode' which works differently from the normal streaming mode. A button up event (or joystick returned to home position) requests the sender thread to cancel realtime jogging mode.

The realtime jogging mode basically does:

while we haven't been asked to cancel jogging {
    Send a jog command with the requested feed rate and a distance based on the algorithm described in issue #741 .
    Wait for OK.
}
Send the cancel jogging realtime command followed by G4P0
Wait for OK.
Return to normal sending mode

The loop quickly fills the buffer to capacity, but the cancel jogging command flushes it. It's necessary to send a G4 and wait for it to complete before you send anything else or weird things can happen (see earlier threads).

breiler, 2019-10-24:
Thanks @AlexHolden, yes this makes a lot of sense and I understand how this is by design in GRBL (the penny now dropped). I had my suspicions that this would be the case and you confirmed it.

I now need to figure out which alternative is the best looking forward, so it this could be reused for other controllers such as Marlin, TinyG and g2core and also the most intuitive for the user.

And thanks @zeevy and @bdurbrow for the other alternatives.

closed 2019-10-24 by breiler

MAK-1966, 2022-02-21:
hello dear friends,

I have a question ; why there is no dedicated JOG pin in GRBL ?
isn't your solution like showing your left ear with right hand 

why GRBL doesnt solve this issue in software ? 
for example a dedicated jog pin like home pin

thank you