gnea/grbl #747: Implement Button Press in Gcode to synchronize Camera Slider with 3d Printer

aviator93, 2019-10-24:
Hello

I run a CNC Shield V3 and an Arduino Uno with GBRL and control it by Universal Gcode Sender. This works so far.

I would like to create time lapses of 3d prints, therefore I want my slider to move only for small steps when a button is pressed. (The button is pressed by the printer head moving out of the way each time a photo is taken.) This the slider I use: https://github.com/Adamslab/OpenSlider

How could I implement this in my .gcode file for controlling the slider motion?

For example:
button is pressed, starts the following
G4 P5 /wait 5 seconds for the camera to capture the image (controlled seperately)
G21 G91 G1 X1 F50 /move the camera slider 1 mm to the right
G4 3600 /wait for next button press (I would use 1 hour here, as each layer takes a different amount of time, maximum around 1 hour)

Thanks in advance
Dominik

bdurbrow, 2019-10-25:
There's no way to do this with Grbl on an Uno by itself. You can either a) write an Arduino program that does not use Grbl; but just uses one of the Arduino stepper libraries, or b) write a program that runs on some outboard controller that waits for the button press and then sends the required G-code commands to the Arduino running Grbl.

It's not really possible to add anything to the ATMega328 version of Grbl anymore - it uses literally _all_ of the flash space on the chip; there's no room to add any code. However, there is a version of Grbl for the Arduno Mega 2560; and there is plenty of room there (I have a fork of Grbl-Mega that adds support for common 3D-printer hardware for a UI & SD card support - at my last compile, the Arduino IDE reported that I was using up 27% of the available flash space).

I suppose that this could be added to my fork... I've got preliminary support for an interface for adding custom M codes... also, M30 would need to be re-configured to "rewind" the G-code file on the SD card.

Hmmm......

aviator93, 2019-10-25:
thanks for your quick reply. I will try to write a program and send the g-code commands directly to the Arduino.