gnea/grbl #748: G30 cause the machine to move on three axis simultaneously

htaxil, 2019-10-25:
In the Commands documentation, we can read for $H command : `So then after '$H' homing, you could just enter 'G28' (or 'G30') and it'll move there auto-magically. In general, I would just move the XY axis to the center and leave the Z-axis up. This ensures that there isn't a chance the tool in the spindle will interfere and that it doesn't catch on anything.`

But in my case, G30 cause the machine to move on the three-axis simultaneously.
Is there any parameter to set somewhere ?

chamnit, 2019-10-25:
Yes. You set the G28/30 positions with the G28.1 and G30.1 commands. You issue them when your current location is where you want it.

htaxil, 2019-10-25:
I set the position well with G30.1
The problem is that the machine moves to these position by moving on all three axes simultaneously, whereas it should move in XY first, then in Z

langwadt, 2019-10-25:
http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G30-G30_1