gnea/grbl #749: Normal probing operations on industrial machines re Door ajar

MeJasonT, 2019-10-26:
Is it normal practice to disable jogging whilst the machine doors are open.
the door ajar command locks out the control of grbl.
If i were using a £600+ 3d digital probe i wouldn't want to jog it up to my work piece to probe zeros looking through a milky oil covered window.
what happens on industrial machines.
I imagine there must be a forced override.
is the issue that grbl is trying to remember its position based on feed hold based door override.
is it possible for grbl to know it was idle and therefore ok to jog or mid job where it wouldn't be ok to jog.

bdurbrow, 2019-10-26:
Most machines have a door override; for GRBL I’d implement it with a 2-way DPDT toggle switch and a nice, big, bright red self-blinking LED. In the normal, non-override position, the door micro-switch switch goes to the Arduino and the LED is unpowered. Flip the toggle to override, and the micro-switch is either shorted (for a normally closed setup) or disconnected (for a normally open setup); and the LED is powered to remind you of what state you left it in.

terjeio, 2019-10-26:
For my CO2 laser I added a setting, ignore door when idle - this to allow me to jog while the lid is open. ( I have a red laser point for aiming I want to see unhindered). I agee that the standard implementation is a bit strict and not always practical.

MeJasonT, 2019-10-26:
Thanks bdurbrow that's a good suggestion i'm going to keep it in hand as a back up to terjeio's idea.

terjeios what did you do to achieve your method, is it a change in grbl firmware.

terjeio, 2019-10-26:
@MeJasonT 
It is, and has to be, a change in the firmware. Perhaps it is possible to backport from my [fork/rewrite](https://github.com/terjeio/grblHAL), but then the only to the Mega version? Be aware that my fork/rewrite is _not_ suited for 8-bit processors...

MeJasonT, 2019-10-27:
Ok i,m still using an ATMEGA 328 based board, tank you very much for your response but it looks like the switch will be the way to go.
many thanks
J.

MeJasonT, 2019-11-01:
Thanks guys,
I installed a key switch which when activated enables a panel lamp and an auxiliary relay.
the relay opens the circuit to the door switch and also disables the 24v feed to the main spindle contactor.  So now i cant destroy a heimar like John at NYCCNC
https://www.youtube.com/watch?v=V8tQsXszzm8
obviously Tormach don't have such a safety feature.
My machine is evolving quite nicely - just want a Haas VF1 sticker to put on it now - just for giggles.

[image]

you can see the controls on the left hand side of the cabinet.
EStop is a total system kill it isolates the mains to both the inverter and stepper motor power supply.

Almost finished the panelling, just have to reinstall the mist coolant and connect up the flood coolant and start making chips again.

closed 2019-11-03 by MeJasonT