gnea/grbl #751: Setting Up Issues

PCPayne, 2019-10-31:
Hi Everyone i have NEMA 17 Stepper Motors using a m8 threaded rod.

What should my settings be? 

I currently have this?

$0=10
$1=100
$2=0
$3=10
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=250.000
$26=100
$27=5.000
$30=1000
$31=0
$32=1
$100=42.000
$101=40.000
$102=0.000
$110=2000.000
$111=2000.000
$112=0.000
$120=500.000
$121=500.000
$122=0.000
$130=600.000
$131=600.000
$132=-0.000

3RD-Dimension, 2019-10-31:
You say threaded rod, are you using a 8mm leadscrew or a threaded rod?
For an 8mm Leadscrew depending which axis you have it on - Steps per/mm is roughly 800 (for setting $100 (x), $101 (y), $102 (z) - that should get you close to the right steps then fine tune if need to.  For max rate mm/min - for nema17 start off with 800 for $110, $111, $112 (x,y,z) and fine tune if need to and look into the other settings $120, $121, $122 (acceleration).

MeJasonT, 2019-10-31:
There are a few calculators available like
http://swarfer.co.za/cnc/grblcalc.html

3rd dimensions advice is a good conservative start and will definitely get you off to a good start

PCPayne, 2019-10-31:
@3RD-Dimension im using a leadscrew

MeJasonT, 2019-10-31:
with the lead screw the pitch of the thread will be more opened out and will travel further for one revolution than a threaded bar

PCPayne, 2019-10-31:
Im pretty new to all this, so thankyou for your help. What sort of settings should i be looking at then? I looked at the site you mentioned @MeJasonT but tbh i got lost looking at it.

PCPayne, 2019-10-31:
https://www.ebay.co.uk/itm/3D-Printer-T8-Trapezoidal-Lead-Screw-Rod-W-Nut-Coupling-Shaft-Bearing-Mount-Set/152604546903?ssPageName=STRK%3AMEBIDX%3AIT&var=452047099192&_trksid=p2057872.m2749.l2649

This is the lead screw with Nema 17 Stepper Motor

MeJasonT, 2019-10-31:
The green bit at the top for Z axis (ignore what the axis name is, its the only part of the calculator using a lead screw the other axis is the table represent a belt driven axis.
most likely your stepper motor will have 200 steps per revolution
the microsteps will depend on your controllers ability to set up the motor to microstep usually jumpers next next to the stepper drivers on the control board.
to get going dont have  set it to microstep then using the calculator choose 1 No microsteps
the Z values are at the bottom of the calculator.

3RD-Dimension, 2019-10-31:
Addition to my post above using 800, that is for 8mm lead and 32 microstepping - forgot to mention that.

PCPayne, 2019-10-31:
@MeJasonT sorry im still lost.

PCPayne, 2019-10-31:
Does this look right?

$$
$0=10
$1=0
$2=0
$3=10
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=250.000
$26=100
$27=5.000
$30=1000
$31=0
$32=1
$100=800.000
$101=800.000
$102=0.000
$110=500.000
$111=500.000
$112=0.000
$120=500.000
$121=500.000
$122=0.000
$130=600.000
$131=600.000
$132=-0.000
ok

PCPayne, 2019-10-31:
Also what settings would i look at for homing?
I have them connected for NO Switches.

When i try the home nothing moves.

MeJasonT, 2019-10-31:
first things first
what controller and GUI are you using

PCPayne, 2019-10-31:
Im using KKmoon controller  & LightBurn Program:

This is the closest link to the one i have:
https://www.amazon.co.uk/HUKOER-Upgraded-Engraving-Controller-V3-3-150W/dp/B07PT93SM8/ref=pd_sbs_60_1/262-9789227-0849815?_encoding=UTF8&pd_rd_i=B07PT93SM8&pd_rd_r=60d87a5c-e4dd-4749-b605-073783567d1f&pd_rd_w=rMWGB&pd_rd_wg=RCq2H&pf_rd_p=cc188cba-1892-42b3-956f-6c67d0ab7a00&pf_rd_r=W71941457RPR292KXRWD&psc=1&refRID=W71941457RPR292KXRWD

MeJasonT, 2019-10-31:
i cant find a copy of the instructions for that board, try looking in your instructions for connecting motor and microstepping if its an option.  alternatively make $100 =100 (if its x axis) at start from there.
if all axis are using same lead screw then make $101 & $102 =100 as well.

so you have enabled the homing setting and have endstops attached to your machine.
$21=1 $22=1
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

MeJasonT, 2019-10-31:
when you get the axis moving then setting the correct steps/mm is perfected during calibration. you will soon find out how many steps you need to travel 1mm etc

PCPayne, 2019-10-31:
Thankyou @MeJasonT ill give it a go, and come back to you.

MeJasonT, 2019-10-31:
just a heads up, get ready to stop it whenever you first get it to move.
when you press home the machine is locked in to performing the homing routine
if your motors run backwards or your end stops are at the min travel position ie inverted
you will scare yourself very quickly.  be in control of that motor power switch.
if all goes well this comment will just be a bad dream.

PCPayne, 2019-10-31:
Ok new issue, motors are now running fine. BUT

I have a circle that i want to be 100mm but it cuts it at 500mm using LightBurn any ideas?

MeJasonT, 2019-10-31:
you need to set up your steps/mm
so if you used 100 change it to 20 
size you got 500/100 size you wanted =5  your steps/mm were 100  divide by 5 =20 (new steps/mm)
there is a calculation to work it out accurately but i cant lay my hands on it just yet - busy at work.
google grbl steps/mm calibration and see if you can find the equation if not i will try and find/send later.

PCPayne, 2019-10-31:
Thankyou, do you know what setting it would be at all? $??=??

MeJasonT, 2019-10-31:
which axis are you testing ?
X= $100
Y= $101
Z=$102
we cant really offer you and instant value that will get you working as the dynamics of your machine will differ from other machines.
If you don't power up your laser but simply simulate cutting a circle - do you manage to reduce its size from 500 mm that you said you were able to do before.
if changing the value to 20 has enabled you to do this then place a ruler at 0mm under your laser point move your axis 50mm and see where it ends.
you can then calculate the difference required to set up the correct steps/mm of just increase and decrease the value until the axis moves just 50 mm using trial and error - much more fun.

The math
new_steps = old_steps * (100 / distance_actually_moved) 
so say it was told to move 100 set at 20 steps/mm
and actually moved 97 for example
then  20 (old steps/mm) * 100 / 97(distance actually moved)  = 20.618 max 3 decimal places ? (need to check might only handle two)  make the tested axis steps/mm 20.618 and it should now move 100 mm.

3RD-Dimension, 2019-11-01:
Handles up to 3 decimal places so can really fine tune it.

MeJasonT, 2019-11-01:
Cheers 3rd dimension.  much appreciated.

PCPayne, 2019-11-02:
Hey guys, i have changed settings $100/101 & 102 to 20.618

And now the motors dont seem to move at all :/

PCPayne, 2019-11-02:
[image]

PCPayne, 2019-11-02:
@MeJasonT @3RD-Dimension id appreciate it if we can get it going please :)

3RD-Dimension, 2019-11-02:
The 20.618 was just an example that MeJasonT gave.

Ok, change the setting to something like 500 first, 20 is just too small of a step count...  

Now, leave the laser off, move the axis that you want to change to a safe point and then mark where it is with a cross or point or even a bit of tape.

Now tell it to move that particular axis say 50mm or 100mm and then measure with a ruler from the point you marked before to where it has stopped now.

Now do the following to work out the steps.
Current_Steps * (distance_wanted / distance_measured)

Example lets say your current step setting is 800, you wanted it to move 100mm but you measured 97..
Then it would be 800 * (100/97) = 824.742

Do you know what microstepping for the stepper drivers you are using? 8, 16, 32 etc?

PCPayne, 2019-11-02:
Which setting from the screen shot am i changing? Sorry i am a complete newbie.

PCPayne, 2019-11-02:
And no sorry not sure what the motors are apart from nema 17
Model No is: 17HS1352-P41

PCPayne, 2019-11-02:
Ok guys update,
Now have the motors running. I have draw a 10mm square. And it burns it at 0.5mm square.

these are my settings:
$$
$0=20
$1=0
$2=0
$3=10
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=250.000
$26=100
$27=5.000
$30=1000
$31=0
$32=1
$100=200.000
$101=200.000
$102=42.000
$110=600.000
$111=10000.000
$112=800.000
$120=10000.000
$121=400.000
$122=500.000
$130=600.000
$131=600.000
$132=600.000
ok

MeJasonT, 2019-11-02:
X steps per mm ($100) = 824.742 if that's the right value ?
(earlier you stated it gave you a 500 mm circle.)
switch the bloody laser off or it will cost you a fortune in material

if you still get stuck enter 200 into $100 and then increase (or decrease) by 50 until your travel gets closer to what it should be then change your increment value to 10 and add or take that away from $100 until you get closer again

do same for other axis, be careful with z it will travel differently (possibly)
once all are near enough concentrate on getting the rest of your machine set up and running
when it works consistently - then make final adjustments to calibrate steps/mm

your motors are moving and i imagine they are travelling in the right directions
the hard work is done, many people don't even get the motors turning first time
just keep playing with the steps/mm you should be just about there.

Just out of curiosity do your motors scream like holy hell then move erratic distances.
If so then the accelerations are too fast.  Thankfully you haven't mentioned anything yet so we are just working on the information you have provided.

PCPayne, 2019-11-02:
@MeJasonT Thankyou so much for all your help. I think we are nearly there.as its scrap wood 

I entered $101=824.742.

I drew a 10mm square and its come out at 20mm square. Would you say i half the 824.742 ??

Also no scream from the motors either. All go in correct direction. 
It does do it very quickly. I dont mind about material as its scrap wood.

MeJasonT, 2019-11-02:
Yep try half
now you get the idea
you will have to post an image of the Graffiti you have produced on your scrap board on completion
could be like a Banksy and worth a fortune.

No screaming is good, the other issue which causes screaming is marriage, especially after working on your machine till 3am.

3RD-Dimension, 2019-11-03:
Thats great news its starting to work - its a lot of trial and error going back and forth changing settings then testing but you will get it exact in the end and the more you change, measure and test you will get it right on the mark.

PS. Don't worry about being a Newbe, its all about learning and we all started there at one time.

PCPayne, 2019-11-03:
@MeJasonT You are a star thankyou so much, my sizes are now correct :) After i halved it.

Now haha

How would i go about setting up homing?
I have switches set up on each side of the axis and connected.

MeJasonT, 2019-11-03:
before you do anything get ready to hit the reset button or stop your machine
this is where some machines start to run weirdly.
read all of this message and hopefully you will get an oversight of what to expect before you hit that home button.

Your machine from power up will be locked in software

at this point you can override the lock by pressing your unlock button or sending $X
don't do this and don't get in a habit of doing this - always home your machine - be disciplined like you are using a pro machine like a Haas or Tormach.

To get going
press your home button or send $H

the z axis shout start to raise (if homing direction is not inverted)
for reference - you have both an operating direction in settings and a homing direction 
GRBL by default raises the Z axis first clear of the work area (for obvious reasons, your work pallet or a vice could be on the bed of your machine)  It will the simultaneously move X&Y axis 
in most cases the home position is set to the furthest right corner of your bed as you face your machine ie it brings the bed forward towards you and pushes the bed to the left.
The writers of GRBL have worked very closely with the GCode standard and the principles and operation of professional/industrial machines.  What you might think of as odd is there for good tried and tested reasons be it safety or a disciplined way of operating a machine.

Industrial machines compute and are setup in negative space, when you look at the feedback of your homing moves you will see for example -300 ,-85, -50 under the axis displays.  Just accept this unless you really want to understand machine principles.  

If you have approached machining/CAM from a CAD or 3D printing environment then you will inevitably ask the question about your work position and why your machine doesn't use the bottom left corner of the work area (instead of the top right)as if its A4 paper.
hence negative coordinates.  You also need to change your mindset from adding material to taking it away.

if you have your drawing printed out on A4 paper rotate it 180 degrees and place the corner of your paper under the home position - that's how your machine will make it.

you can invert the axis later so that it burns/mills/prints
as if it were drawing on A4 paper if you wish but then your dream of operating a Haas has just become a hobby thing.  I like the idea of sticking to the plan and learning how to do it all properly.
it also makes problem solving and asking questions a whole lot easier - i tried to fight the system when i first started, you soon find out this is a forum and not a church - there is no forgiveness here.
Pleased to say your doing great bud.

best of luck
J

PCPayne, 2019-11-03:
Ok i think i understand this:

I have these following coordinates:

START LOCATION=
X: -170
Y: -150

END LOCATION=
X: 210
Y: 30

WORKING AREA:
X 380
Y 180

Is there any particular settings i should change before hand?

$$
$0=20
$1=0
$2=0
$3=8
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=0
$24=25.000
$25=250.000
$26=100
$27=1.000
$30=1000
$31=0
$32=1
$100=412.371
$101=412.371
$102=412.371
$110=600.000
$111=500.000
$112=500.000
$120=500.000
$121=400.000
$122=400.000
$130=400.000
$131=200.000
$132=100.000
ok

PCPayne, 2019-11-03:
Ok so i tried it, motors sounded like they were kicking in but didnt move.

PCPayne, 2019-11-03:
Done a quick google - How do i disable Z homing? As i dont use a z axis. I only use X & Y.

MeJasonT, 2019-11-03:
ok i was writing a reply when your last response came in - looks like we are working live in real time.
I wanted to ask if you had the settings correct for homing -yes you do
then started looking for the speeds for seek and homing in the settings - again looks like you have
you need to 
set your machine limits
$ 130,$131,$132 your bed travel dimensions
notice i said travel and not the actual dimensions

ok looking into z for you

MeJasonT, 2019-11-03:
when your homed press your axis (0) zero keys for X & Y

we are going to disable Z but normally z zero is down at your work height

PCPayne, 2019-11-03:
[image]

Which key do i press when i press home? It looks like i have Z disabled.

PCPayne, 2019-11-03:
Im getting ALARM 8 if that helps.

MeJasonT, 2019-11-03:
get rid of the dirty big pop up window i cant see lol.
i realise you are pointing out that z axis is disabled.
i'm nor familiar with your GUI so if you close the window i'm looking for something that looks like a home button please send new image.

MeJasonT, 2019-11-03:
Alarm 8
try adjusting your home pull off setting $27
say to 8 (8mm)

PCPayne, 2019-11-03:
[image]

Tried changing that setting, still alarm 8 after a few seconds.

MeJasonT, 2019-11-03:
yep answer to earlier question home button has a house on it
bottom left of image.

PCPayne, 2019-11-03:
Yes ive tried that once pressed, motors click but then alarm 8 in console.

MeJasonT, 2019-11-03:
Ok i pulled up alarm 8
https://github.com/gnea/grbl/issues/563

MeJasonT, 2019-11-03:
still looking for alarm list, im old and slow not like you young pentiums you know

PCPayne, 2019-11-03:
Haha thanks Jason, ill try changing $27 and see if anything changes.

MeJasonT, 2019-11-03:
power down your machine move you axis so they are no where near home position then power up again.

basically it cant move away from home as its already triggered
you have made a change to the pull off but grbl may has required a refresh
performing the action above will achieve both

PCPayne, 2019-11-03:
Its already located in the center, no where near switches. Would the pins need inverting?
[image]

MeJasonT, 2019-11-03:
what happens is the machine will travel until the end stop is triggered then will move away by the $27 value go back and seek the end stop again then pull off the switch to clear it.

im thinking your disable z does not work with the homing command, we may need to search for the answer for the laser software/frontend ? how do others with lasers achieve it

there is a code alteration in config.h file which sorts this out in grbl but like you have said your no coder - it will take a while to get you messing with dragons.

PCPayne, 2019-11-03:
Yes ie seen that alo but as its not a arduino i cant access the config.h file to do it.

MeJasonT, 2019-11-03:
you are using grbl firmware

MeJasonT, 2019-11-03:
did you load the control board with code yourself

MeJasonT, 2019-11-03:
for reference
these are the links i have open which im using to assist you
http://domoticx.com/cnc-machine-grbl-error-list/
https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#how-do-i-configure-my-homing-cycle-to-just-do-2-axes-or-i-have-a-pen-plotter-and-dont-have-a-z-axis
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

PCPayne, 2019-11-03:
No it came pre loaded with grbl

PCPayne, 2019-11-03:
Ill download the arduino software and see if it picks the controller up.

[15 more replies]