gnea/grbl #752: TRIGGERED_MOVEMENT option

FelisucoVFR, 2019-10-31:
This is just a request, not a bug but I don't see the option to tag it as a request or question. Sorry.

For my actual project, it would be great to have the TRIGGERED_MOVEMENT option, same as teathimble version has.
The idea is not processing the next gcode line until receiving an external trigger.

I have found this very used at DYI embroidery machines but only found the teathimble reduced option.

Thanks

langwadt, 2019-11-01:
just change the code, could be a simple hack of G4

MeJasonT, 2019-11-01:
umm, looks like you want a physical switch to trigger a single step gcode send to grbl.
that would require a reprogram of the firmware to use a pin like A4 to to trigger a single step, unfortunately the gode in grbl is buffered and sits in a block of a few commands. so the way it executes those commands would need altered as well.
langwadt's suggestion would work but involve the user send a resume command from the keyboard after each line send.
operating software like grbl-panel allows for sending single step instructions but would also require single step triggering from the keyboard.

FelisucoVFR, 2019-11-01:
@langwadt thanks, but that is not the behaviour I need.
@MeJasonT Correct. I am just code friend, not expert. In the teathimble code they still use buffer but the Arduino wait for the external trigger if triggered_movement is defined. I am sure for any code writer taking a look to Teathimble code would take a little to add to grbl.
Anyway it is just to not have lots of firmwares and keep all my projects under grbl with the several advantages. Worst case I can use Teathimble.

langwadt, 2019-11-02:
why would modified G4 need user input? modify the code for G4 so that a "magic"  value waits for a pin instead of time, add it to your gcode whereever you want to wait for a pin

MeJasonT, 2019-11-02:
looks like your best bet is to buy an old 3d printer marlin board and load it with the teathimble firmware

FelisucoVFR, 2019-11-02:
@langwadt That sounds great, but I am not skilled enought to know how to modify the G4 behaviour. Could you guide me for that?

@MeJasonT I have tried it in a Mini Pro and works nice. It is just because I like sharing one firmware for all my cnc’s with just different settings instead of one different for each. Just that and the way grbl works with the accel, jerk and other movements which I prefer. I thought that the add could be easy for a clear mind software writer, not my case... just Let’s say trained beginner.

bdurbrow, 2019-11-03:
I'd suggest considering moving to an Arduino Mega, instead of something based on the '328. The reason is there's barely any flash storage left on the '328 after all of Grbl gets loaded; however there's plenty of room on the ATMega2560. I have a version that adds UI functionality via common 3D printer parts; with all the stuff that I've added I'm still only at about 27% flash used.

Adding a wait-on-external trigger M code wouldn't be hard; and I've also got hooks for doing that if you are using my fork of Grbl-Mega (have a look at `grbl_expansion_interface.h` in my github repository - note that this is somewhat preliminary; so if you do choose to try it, and have any problems please **do** open an issue on my fork's github repository).

The official version of Grbl-Mega: https://github.com/gnea/grbl-Mega
My fork: https://github.com/bdurbrow/grbl-Mega