gnea/grbl #756: Cycle start not responding after feedhold

SketchThis, 2019-11-06:
As the title says I have hardware buttons for e stop, feed hold and cycle start wired up. They all function as described in the docs.... However after I do a feedhold I can't use cycle start to get the job going again... I had an older GRBL setup that worked this way... 

So why can't I use the cycle start hardware button after a feed hold?

hairu, 2019-11-24:
If Grbl leave the HOLD state and go IDLE after you send a cycle start real-time command then you may want to check that your sender is still sending lines to Grbl.  If it stays in HOLD then check that there's not a logic bug in your sender that is preventing the cycle start real-time command from being sent.

abdset, 2019-11-29:
In grbl source code, config.h file, i found this:

// This option causes the feed hold input to act as a safety door switch. A safety door, when triggered,
// immediately forces a feed hold and then safely de-energizes the machine. Resuming is blocked until
// the safety door is re-engaged. When it is, Grbl will re-energize the machine and then resume on the
// previous tool path, as if nothing happened.
// #define ENABLE_SAFETY_DOOR_INPUT_PIN // Default disabled. Uncomment to enable.

If you are using a safety door and is enable, you may close it back to resume the feed.