gnea/grbl #757: G28 / G30 Homing error:9 - why ?

AndKe, 2019-11-06:
```
Grbl 1.1h ['$' for help]
[MSG:'$H'|'$X' to unlock]
<Alarm|WPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=8
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=800.000
$101=800.000
$102=800.000
$110=800.000
$111=800.000
$112=800.000
$120=10.000
$121=10.000
$122=10.000
$130=320.000
$131=140.000
$132=34.000
ok
G28
error:9

 ```

I do have endstops on every axis, and based on that I can move axis with "offline controller" , it should be fine. (none of them falsely activated)

3RD-Dimension, 2019-11-06:
Error 9 points to it could not find limit switch within search distances. You can try increasing max travel ($130, $131, $132) and/or decreasing pull-off distance ($27).  Also check your wiring.

neilferreri, 2019-11-06:
You cannot send gcode (G28), while in an alarm state. To home you'll send $H, not G28. The G28 position is set by sending G28.1 and is only valid when homing is enabled.

AndKe, 2019-11-06:
Thank you, needed $H , this is so much different from the gcode I know from Marlin :)

closed 2019-11-06 by AndKe