gnea/grbl #758: Steppers do not move after upgrade to grbl 1.1

iivoilov, 2019-11-09:
My CNC machine came about 1 year ago but I started to use it now, It came with grbl 0.8 and was working. When I started to use it I run into the problem that after running to the first position everything is shifted the same way, i.e. it looks that the 1st move is done always shorter than it should be but further moves are correct. When I issue commands manually in terminal this problem does not appear, all moves are correct. I tried to reduce max rate and max acceleration but it did not help.

So I suspected that such problem is solved in the latest version of grbl. I installed it on a new Arduino Nano and replaced the it in CNC machine. Got a proper prompt "< Grbl 1.1h ['$' for help]", put the same $$ settings like on the old grbl 0.8, but unfortunately only spindle is working. Steppers do not move.  I played with $0 value - no luck.

Please advise which settings may affect steppers

I added image of the control board of my CNC machine
[image]

iivoilov, 2019-11-09:
Arduino Nano on my board has CH340G chip which has the known issue. Does this issue appears in such way?

iivoilov, 2019-11-10:
Found the problem. The board has different wiring than default CPU_MAP_ATMEGA328P.
So I changed XYZ step and direction pins, recompiled and uploaded the new version of grbl.
Checked the problem with short move on the first motion - gone.

closed 2019-11-10 by iivoilov

shreekant-sureliya, 2020-10-14:
i also tried this and it worked thanks for info.