gnea/grbl #759: cnc plasma freeze at point of Fire

pedrotorres25, 2019-11-10:
I have built my first very basic cnc plasma. The cut file, using grbl, frezes at the point the relay fires the plasma on. The same file runs through fine in its entireity without the plasma on.
I installed a sparkfunk relay with a led and the led lights up at points the plasma should fire on all test runs while the plasma is off. The same cut files with plasma on the relay fires the led comes on the plasma air is released and a spark then freeze. The following happens after freeze. The spark is gone two seconds later the air out of plasma torch stops and the led on relay goes off. The UGS is frozen. I disconect controller from pc and unplug power pack restart UGS then plug power and controller back to continue trial and error.

I have changed spindle speed. Anything less than 800 does not freeze the file it runs as if the plasma is not connected. Spindle speed above 900 gives the spark and freeze. I did not try any speed between 800 and 900.

FelisucoVFR, 2019-11-10:
@pedrotorres25 That is noise entering to the controller.
Faced before with a small CNC
I think you have to think on optocouplers and capacitors at all signals connected to the controllers.
Check first if the relays are optocoupled. That would be first noise source. Then the end of track switches. Try to use small capacitors to reduce noise. About values better use google as I did.
If cables are long or pass near the plasma, change them by shielded cable too.

MeJasonT, 2019-11-10:
all of the above plus.
what sort of plasma are you using - High frequency or scratch start.
the latter requires a pilot arc to be generated that once energised will keep the conduction between tip and stock going.  The high frequency type generate a burst of energy which creates a charged atmosphere between the tip and the work piece.
The reason the cheap ones are called scratch start is without a pilot arc the only way to create the plasma flame is the earth the tip against the work then withdraw to a cutting height of 1.5mm ish. depending on thickness of material and amperage.
if you dont get an arc started its like throwing your grbl controller into an electrical power plant.  loads of lovely high frequency noise, guess what it trips.

pedrotorres25, 2019-11-10:
Thank you for quick response.

I am using a cheap zenny cut 50 plasma with pilot arc.  It is blowing a hole through the material but quickly freezes.
The xmotor wire and plasma cable are touching.  I will turn the x motor 180 degrees to see if there is any difference. I think the relay wire are far enough away from everything else, but I will work on the noise on the signal side of relay as the next step.
I will update my progress as soon as I can.

bdurbrow, 2019-11-11:
A quick look at an amazon page indicates that this is not a HF-start model. So, I'm suspecting that you are dealing with regular electromagnetic interference caused by running high power cables next to cables that connect to the microcontroller.

Suggestions:

* Re-route cables for the plasma cutter itself & torch away from other cables.
* Get some braided metal wire mesh sleeving and put it over the cables. Connect the wire mesh to ground.
* Put the processor running Grbl into a closed, grounded metal box.
* Add ferrite noise chokes to the cables.
* Consider adding small capacitors across certain pins where they enter the box for the Grbl processor. The pins that you are looking for are any signal pins; and you want to put the capacitor between the pin and ground.

MeJasonT, 2019-11-11:
For ref
[High frequency noise EMI.docx](https://github.com/gnea/grbl/files/3832785/High.frequency.noise.EMI.docx)
[Linear Regulated USB Isolator.docx](https://github.com/gnea/grbl/files/3832786/Linear.Regulated.USB.Isolator.docx)

pedrotorres25, 2019-11-11:
Thank for all the help so far.  I was lost but now I have some guidance.  I made sure everything is grounded and let the cnc work as a test run and use the plasama torch by hand nearby and sure enough freezes while I use the torch nearby.
I have purchased on amazon all that bdurbrow sugested and now just waiting for delivery.  
I do not have ferrite nose check at any cable.   Purchased
I do not have braided metal mesh over the plasma cable.  Purchased
Also will make adjustments on the metal box and also purchased capacitors and resistors for filtering noise.

I will update my progress

bdurbrow, 2019-11-11:
What type and value capacitors did you order? Or did you get an assortment? (I just want to make sure you have the right thing). 😄

pedrotorres25, 2019-11-11:
E-Projects 100EP5124K70 4.7k Ohm Resistors, 1/2 W, 5% (Pack of 100)

OCR 24Value 500pcs Electrolytic Capacitor Assortment Box Kit Range 0.1uF－1000uF

Cedmon 20 Pieces Clip-on Ferrite Ring Core RFI EMI Noise Suppressor Cable Clip for 3mm/ 5mm/ 7mm/ 9mm/ 13mm Diameter Cable, Black

Electriduct 1" Stainless Steel Braided Sleeving (304SS) - Length: 10 Feet

bdurbrow, 2019-11-12:
I was afraid of that. The capacitors you got are way too large for use on signal lines (they are still useful for power lines, though).

You should get an assortment of ceramic capacitors in the range of 0.01uf to 1uf... and use the smallest one on any given line that you can get away with.

The rest of it looks OK, though.

pedrotorres25, 2019-11-12:
I just added an assortment of ceramic disc capacitors for the signal lines.

pedrotorres25, 2019-11-13:
I had done some more reasearch as I wait for my purchases to arrive.  I did fail to mention I am using an arduino Uno and for stepper drivers TB6560.  I do believe the drivers isolate 5v from the 24v.  The beefcake relay I am using just for the fire button of the plasma torch.  The plasma itself is on its own 20 amp outlet.  The laptop and compressor and power supply are on the same outlet.
The wire between the beefcake relay and plasma box is short and maybe the plasma box is to close to the arduino and signal wires.  Can that be the root of the problem?

bdurbrow, 2019-11-13:
Although in general it's good to state _exactly_ what hardware and software you are having a problem with; in this particular case the troubleshooting steps are the same for any electronics having similar issues.

> The wire between the beefcake relay and plasma box is short and maybe the plasma box is to close to the arduino and signal wires. Can that be the root of the problem?

Well, strictly speaking, the _root_ of the problem is the fact that the plasma cutting process is electrically "messy", spewing lots of radio waves out into the air.

However - given that received radio power goes down with the square of the distance - just moving things apart can help a lot.

The wire between them can still be an issue, so putting ferrites and capacitors on them can be also necessary.

MeJasonT, 2019-11-13:
@bdurbrow I'm glad someone did the maths bit.
The one thing everybody forgets is extraneous metal surfaces.  Steel framed buildings, steel racks and other machines etc. radio waves and electro magnetic interference love accumulating around such surfaces like a huge capacitor just looking for an earth path (lower potential).  I often describe it like a Hollywood film or Frankenstein where electrical arcs bounce off every surface and travel along pipework.  The interference we are talking about looks just like that only invisible.
I often get issues if the table is moved too close to the mains socket (which is pretty close to the meter and consumer unit) the electrical installation earth is not a good ground In the UK its actually bonded to the neutral line back at the substation in most cases TN-C-S.
https://en.wikipedia.org/wiki/Earthing_system

cri-s, 2019-11-13:
On the CNC interface normally one contact need a relay and the touch
interface (ohmic contact) if you don't have a additional floating z axis
with microswitch the ohmic contact need a 9v battery and resistor, 11mA
should flow.

MeJasonT, 2019-11-13:
@cri-s  are you describing a torch height method ?

What can be done is to lower Z axis so tip touches work and then strike an arc.  When an arc is achieved raise the Z axis to 0.6 to 1mm as a cut height.

another method is using a pilot arc - if you have a scratch start machine you can replace your torch ie PT 31 with a torch which has a built in pilot arc connection.  The operation would then be to move down to Z cut height, trigger the pilot arc relay, energise the arc relay and then disengage the pilot arc.  You would not want to keep the pilot arc on as you would absolutely horse through consumables (tips).  Creating a automatic torch height controller for grbl is almost impossible as grbl does not have the ability to run the auto Z height control or feed commands to a torch height controller.  Expensive plasma cutters have controls built in to control THC's
There is an arduino based THC floating around somewhere on the tinterweb.
Two things effect torch height sensors FLAMES and RF noise.
Picking a sensor that can measure the Z work height mounted to the torch which wont act as a huge aerial is challenging.  I had considered an Infra red beam sensor, i also looked into capacitive sensors which funnily enough are used on pro systems.  Unless you plan to cut on slopes or over corrugated iron sheet then setting a z work height and just cutting (if your using mostly flat sheet then you can get away without needing a height sensor.
But.
If you get your work height wrong you can get some horrendous hf noise generated especially if an arc fails to strike.  Touching and dragging the tip over the work can be problematic but  >1.5mm gap is big enough to stop the arc jumping between the tip and the work surface.  Dirty or rusty metal is a whole new ball game - clean metal and bed slats to get a good connection - always.
Worn consumables start the ball running all over again.

Its a miracle my machine even works

pedrotorres25, 2019-11-13:
Update
I rotated the table 90 degrees and extended the high voltage side of wires from relay to the plasma box fire button.  The plasma box is as far as the gun cable will extend and still do some work.  Diagnoly across the table from the plasma box location are the controls.  I installed almost all ferris beads purchased to every wire coming into the arduino controller and the usb cable from pc to arduino twice. Grounded the table with a wire to copper tube into the dirt.

Result :  The same file test run no problem. The same file with the plasma on did not freeze at the same point it actually cut out the inner hole of the test piece then freeze.  I was celebrating to early when I saw the first cut.  The problem is still there but now I know everthing I purchased is needed.  I will install the braided shield wire to the plasma cable and ground that too. Also thinking of detaching the control box from table to keep the distance from plasma box even greater.  Also, as has been pointed out, I will install the ceramic capacitors to increase the filter.

I dont have a Z axis. For the moment I am manually setting the z height.

Question Bdurbrow:
At the fire point or pilot arc location,  does this arc create a greater problem than the location of the plasma box?  My controls, as of now are very close to the table 0,0 location.  The pilot arc fire location is much closer to the controls than plasma box location.

MeJasonT, 2019-11-14:
@pedrotorres25 
A quick test you can try
Disable hardware limits in settings
Home your machine (it will still seek the hardware switches) then try a cut again
if it still crashes - disconnect the limit switches after homing then try a cut.
the cables which are most exposed to the noise from the torch tend to be the limit switches
the forum is full of message threads with noise issues causing false end stop triggering.
its something that indicated i had a screen issue with my cables.
Also only connect your screens at one end - The end closest to your earth (copper tube) don't make an earth loop its counter productive.
is your copper rod in damp earth - if the ground is too dry it looses its potential, water it with a watering can.

pedrotorres25, 2019-11-14:
Agreed.  Based on what I have seen on other forums I disconected the limits switches and have been testing without them.  I have not been homing but manually pan and reset zero to my choice on the work sheet.  
With what you bring about the copper ground I think I will move it where I can actually sink it further and fill it with water right before testing again.  
I have just finnished snakeing the braided shield around the plasma cable, that was difficult.

pedrotorres25, 2019-11-14:
Update
As it turns out looking closely at the wiring I still had 5v going to the motor drivers that were conected to the limit switches.  I took the 5v wire and placed it on nothing on the breadboard.
amazing diffrence the problem is smaller but still there.  I actually cut out a piece almost entirely until the heat warped the thin sheet and the arc prematurely shut down and the freeze.  But the problem is getting smaller and soon gone.  

I still have not found a wire diagram to plug in capacitors where needed.  I did put them on the breadboard and the result was little movement from motors.  I reconected the signal pins from driver directly to arduino.

I am still waiting for the shielding cable for stepper motors should arrive in two days.  I also took the wire from the relay and routed up and over the gantry and run it the rest of the way with the plasma torch cable and ground also runs together with it.  I did that to keep the loop small.  I know the torch height without a z axis will be a pain but that is for another forum.

I will update once the stepper motor wires are replaced with shielded wires.

bdurbrow, 2019-11-14:
>  I did put them on the breadboard and the result was little movement from motors.

If you put them on the step and direction lines; and there was little movement from the motors, then the value of the cap is too large. Try a smaller value - probably much smaller.

Are your stepper drivers in an external metal box, or are they in the same box as the Arduino?

If they are inside the box, then you should (in theory) only need to put caps on the lines that go outside of this box; and not between the Arduino and the drivers. The box should act as a shield for what's in it... well, that's the _theory_ anyway.

Assuming that the Arduino and stepper drivers are in the same box, lines that are candidates for caps should just be the limit switches, and the trigger output that goes to the relay.

I would suggest - if you haven't already - that you put the power supply for the stepper motors in it's own metal box. Switching power supply modules can sometimes be a bit noisy (this is one reason why many come in a metal box from the factory). The output lines from that power supply module are a good candidate for putting a big electrolytic cap (probably obvious, but do make sure they are in the right orientation polarity-wise; and have the requisite voltage tolerance rating).

The firing point location relative to the electronics can be an issue; but do keep in mind that the entire circuit is emitting radio waves, from the connection on the Cut 50 box, thru the torch lead, to the metal being cut, and back thru the return cable (often refereed to as the "ground" cable; which may or may not be connected to an actual grounding rod, depending on your specific plasma cutter and the local mains wiring situation).

Don't be afraid to try moving anything when hunting for a source of noise & interference - sometimes things can be counter-intuitive.

cri-s, 2019-11-15:
Yes I have described torch height control. there is a third method, rc
servo with 90 or 180 degree rotation movement of arm/stick with
microswitch. It is the cheapest option but need more time as example the
sliding z axis in combination with the Z homing switch.

Basically you don't want grounding the arduino, or better speaking, the
system is already grounded. The voltage sensor is usually not ouptocoupled.
There exist some design that uses U/F converter with optocoupler to isolate
it but normally it is not isolated.
This mean that Ground is already connected to chassis ground, and this is
or can be a different ground as the table ground. If you ground it
separately then you create a ground loop and break the star grounding rule.
The same is the reason hat  the ohmic sensor is optocoupled using 9v
battery and solid state relay that tolerate the voltage requirements in
order to not have grounding loop in addition to the high voltage ignition
problematic. However this is not always true as mostly the voltage sensor
is isolated using relay and only engaged when the ignition OK return line
is high.
Basically the reed sensor that have 3-4 turns of output plasma current line
wrapped around it. This isolate the grounding from voltage sensor and
permit  using the ohmic sensor without being isolated. This relays is
external to the controller and normally user supplied, required only for
high voltage ignition. For contact start ignition this is generally not
required. It eliminate problems in case of insufficient inductor isolation
or when the voltage sensor is taken from the wrong place, for cutters that
are not equipped with CNC interface or eventually if the CNC interface
require the relay at CNC side because of the insufficient inductor
isolation.
This isolation is required for the MOSFETs inside the cutter but some
designs tolerate more spikes as others or are designed at edge and for the
voltage sensor it could be too mutch.
In this case even if the CNC interface is supplied or present or ordinable
as optional the relay for the voltage sensor isolation during
ignition/piercing is required. In addition this relay disengage the
ignition start line when it switches the ignition OK signal on, that is
another relay, either inside the CNC interface or at the handpiece that
overrides or replaces the start button. It is not bad having that relay
(ignition OK)in case the control computer hangs and don't clears the
ignition start signal even if the ignition OK return line is closed. It is
a cheap protection for that fault case and to protect the ADC sensor during
the piercing.  If you want I can give you hex file for the plasma cutting.
It use serial shift register for the ui but it work without problems on
serial connection too, just optocouple it or it have the undoc option for 3
line LCD interface (16*2) with some debug Info's to be enabled with eeprom
setting. You need always the optocoupler for the serial interface, better
to use the rs422 interface and ETH cable if you have a longer need for the
cable.

MeJasonT, 2019-11-15:
@cri-s  I would like to see the code for your torch height control/voltage measurement.  It would be great if you could share.  Do you have a GIT account you can link to.

My approach was to use a resistor/potential divider connected across the lance and ground connections and reducing the voltage down to less than 5v. 
The actual voltage measured at the output was 170v at arc ignition dropping to around 8v when cutting. I figured out how to bypass grbl sending z moves (in the unlikely event during cut) and injecting a new z moves based on a height measurement from a (DIY) capacitive sensor.
The arc/grbl freezing thing is why i ended up not bothering.

Just to tame the elephant in the room
when we talk about plasma machine ground connected to the table, its actually the positive.
we don't need to get tied down in welding processes and theory as it will end up confusing some.
The table slats/cutting surface can and should be connected to the copper earth rod.  Any mass of metal surface will cause rf/emi issues - Its a big aerial which just sucks it up and sits there buzzing like a ******** .

I haven't tried cutting aluminium plate yet but left wondering if i might need to reverse the machines output polarity.

@pedrotorres25  hopefully you don't feel we are hijacking your thread/message but as you can see there is a lot going on behind the scenes of what may appear to be a simple question.
Cutter work height, current, clean metal, electrical noise isolation, wear and tear on consumables, machine setup etc all play a part in the creation of your noise/fault issue.  good news is your doing the right thing and just having a go.  if you take the advice and suggestions on board or not, you can take pride in the fact you have built a machine and it damn near works. (that's like the other 80% of us on here who have built plasma tables)

sinfocomp, 2019-11-15:
[image]
This worked for me and i use a cheap HF china plasma cutter, same capacitors added to the step and direction lines just at the drivers, i use arduino mega and ramps , hope it helps you 
BTW, all recomendations also apply shielding, grounding etc, this was my final solution after reducing the noise just as you did

pedrotorres25, 2019-11-15:
bdurbrow:
Reading through the suggestions I realize with information outside of this forum influenced using the wrong capacitors.  Thanks for pointing that out.  I used 0.47uf for the signal pins to the drivers of motors and realy.  I will try again Following the tip I missed.  Also considerieng the power box as an additional problem because I did put it in the  same box with dirvers and controller.

**You should get an assortment of ceramic capacitors in the range of 0.01uf to 1uf... and use the smallest one on any given line that you can get away with.
**

MeJasonT:  I dont mind the Zaxis torch height posts.  I will read over them when I get to that bridge.

sinfocomp:  Thanks for the Diagram

pedrotorres25, 2019-11-15:
I have now 5 capacitors 0.1uf  (smallest ceramic value recieved) installed on both x and y motors.  Step and Dir signals for both motors and the fifth capacitor at signal for relay.  It works so much better.
The cut path was almost completed and the heat warps the sheet metal a bit and the flame stops but the pilot arc continues.  The x and y continue to work through the plasma sputtering and at wich a full flame comes out freeze.  It was almost done traveling the cut path.  I am begining to think torch height may be the new source of freeze problem.  Going to continue to install shielded  stepper motor cables and capacitors.
I have learned alot here with everyone chipping in.  Special thanks to Bdurbow and MeJasonT.

pedrotorres25, 2019-11-16:
Thanks for all the help pinpointing the problems.  It is now working.  Any tweeking necessary I can handle it from here on.

To any future reader:

I started a build for a 2.5 cnc plasma cutter (2 dimensional cuts and weld into 3d).  At the time it was ready to test first cut,  I had no idea how much RF a plasma creates to disturb lower voltage electronics.  Through out this forum I got lots of help to test and correct.  
My mistakes were many.  The power source for the motors and electronics I placed it all together with just wires coming out to the motors and the power outlet.

I replaced the wires the stepper motors come with by shortening them and splice to shielded wire and used Ferrite beads on almost all wires.  Braided metal shield on the plasma cable.  Added capacitors on all signal lines from arduino to breadboard to dump excess noise to the ground.

Basically most of what is above. 

 **Bdurbrow said it best: 
Don't be afraid to try moving anything when hunting for a source of noise & interference - sometimes things can be counter-intuitive.**

closed 2019-11-16 by pedrotorres25

KhanDankata, 2023-01-01:
> 

Hi, i know its a bit old post but ... Can you give me some more explanation on where did you installed your capacitors and what size you used. Thank you.

KhanDankata, 2023-01-01:
> [image] This worked for me and i use a cheap HF china plasma cutter, same capacitors added to the step and direction lines just at the drivers, i use arduino mega and ramps , hope it helps you BTW, [...]

Can you give me info why are those Diods used for in your scheme? 
Thanks

lalo-uy, 2023-01-01:
When the motors wind get switched off, the inductance develops spickes of
hi voltage. The zenner diodes clipped then to less tham 28v.

KhanDankata, 2023-01-01:
> When the motors wind get switched off, the inductance develops spickes of hi voltage. The zenner diodes clipped then to less tham 28v. El El dom, 1 de ene. de 2023 a la(s) 08:59, KhanDankata < ***@***.***> escribió:
> […](#)

Is this necessary for small motors, i have nema 17 motors- do you think i can do the HF filter without the diods ? 
Thanks!

pedrotorres25, 2023-01-01:
What worked well for me is to look at every possibility.
1 Controller location in a faraday cage (below water table)
2 Shielded cables from nema motors all the way to faraday cage or metal box
3 At the metal box and shielded wire location ... magnet choke
4 power source in a separated metal box from controller (usually
manufactured in a box and ready)  Thje wires from the power source same
care as the wires from stepper nema motors
5 If it works on a dry run and not when you intend to fire the plasma
 Concentrate on protecting the controller and a good ground

Capacitors are the high voltage non polarized
Pedro

KhanDankata, 2023-01-01:
> What worked well for me is to look at every possibility. 1 Controller location in a faraday cage (below water table) 2 Shielded cables from nema motors all the way to faraday cage or metal box 3 At [...]

Bro , thanks a lot for the reply i really did not expect anyone to answer its an 4 years old post - It is so nice thank you.
This may sound like a silly question but where do you ground you drain wire from the cable shields ? Is grounding it to the faraday cage good solution or am i totally wrong ? 
Thanks again!

pedrotorres25, 2023-01-01:
You're plasma torch unit is grounded but the flame on the cutting surface
is not grounded that way

You need to pick one ground location ( the body of the CNC)
And ground everything to the metal body that's of course one way to do it.
Everything is grounded one way (avoid two locations as that creates a
ground loop)
By the ceramic non polarized capacitors scratch the high voltage term I
used earlier

KhanDankata, 2023-01-02:
Thanks again, may I know what controller board are you using ?
Im planing to use this one, but i think ill have problem adding capacitors because there arent ground terminals. what do you think ?
https://www.aliexpress.com/item/1005003246920397.html?spm=a2g0o.cart.0.0.454f38davtPpLO&mp=1