gnea/grbl #761: A question regarding future proofing of Grbl for arduino uno /nano based projects

MeJasonT, 2019-11-10:
I know this question has been asked in a round about way in many differing ways but actually the concept make a lot of sense.
Ignoring the super computer SamD chase for the stars approach, which sound very much like microsoft going from 8086 through 486/586 to the newest quad core processors.
The Grbl linage is starting to do the same.  Must be a programmer thing.

If the services on one arduino uno could be shared with a second co processor (UNO) then the pin count and file size could be drastically reduced.
ie if one board handled alarms, auxiliaries like coolant/probing and facilities for SD card and LCD display then the main UNO processor could handle nothing but the planer control and comms to the other board obviously.  A haas or Fanuc certainly has more than one processing device, there will be a PC to handle the programming and display, a PLC to handle the movement and spindle perhaps another PLC dedicated to a pallet/tool changer, if there is a robot in there it may have a pc interface and a PLC.  So trying to squeeze the whole world onto one IC is asking a huge amount and is certainly very commendable for its creators.  Genius every single one of them.
There have been a few issues raised where supporter of the grbl project have asked for integration with a second board for to address various additional support issues and having a fork of this project to address these requests will certainly aid in keeping the base version of grbl going long term.  It will depend on the developers future plans as to where grbl goes in the future but as you move into more a personal computer/micro pc based project then the cost for the micro and hardware escalates ending the DIY hobby based beginnings of the project for many.  The baby will turn into a monster.  Th raspberry pie is one step towards oblivion, i don't see the point of spending in the region of £150 for a stripped down version of a computer (with Graphics) when i could do exactly the same with a laptop.
The flight simulator guys are really hot on using arduinos now to integrate the whole cockpit experience,  they have the ability to hook up many USB devices to a single computer to produce the realism of a full Boeing 737 cockpit.  So using multiple arduino uno;s to furnish a full cnc machine isn't beyond the bounds of possibility.  Th usability and ease of integration with hardware is what has put the atmel atmega miles ahead of microchip PIC's.

I asked a question three years ago about the Sam processors as there was little in the way of hardware for budding diy project builders to get there teeth into, that has still not changed - there is little or no reference or wiki suggesting how one might build there own cnc machine based on a Sam processor. Only Smoothie a commercially based product.

langwadt, 2019-11-10:
why waste time and effort trying to get multiple AVRs to communicate, when there are numerous ARMs with 10x the memory, horse power and pins for cheaper?

MeJasonT, 2019-11-11:
cos 5v tolerant is cheaper and works in a more friendly manner perhaps

cri-s, 2019-11-11:
Having a arduino with sd card and lcd that drives the grbl and uses
software rs232 is
simple to do and code was posted here https://github.com/grbl/grbl/issues/623 .
https://wiki.shapeoko.com/index.php/XLCD  is such a project, there
exist others using
arduino nano.

For grbl itself, adding sd card support is a bit problematic because
of the needed code space, instead using i2c memory cards
https://robotdyn.com/eeprom-data-module-at24c256.html or soldered i2c
memory is unproblematic. Further adding text lcd module
is no problem. I2C adds approx 800 byte to the code and the lcd adds
550 byte to the code.

Adding a shift register or i2c expander is not a problem.

bdurbrow, 2019-11-11:
This is only a problem on the ATMega328; and the constraints are flash space and IO pins.

The Mega 2560 version has no problem running both the Grbl g-code interpreter & motion engine, as well as a UI with SD card support. Clone Mega 2560 boards can be had for less than $10USD; a RAMPS 1.4 is about $4USD, basic hardware for the UI will run you another $12USD or so; deluxe hardware (add an MPG, cycle switches, and override pots) will add on about another $25USD.

Getting two or more '328s communicating and behaving in a fail-safe and synchronous manner is a non-trivial problem; and will eat flash & IO pins. Also, you won't really get improved performance out of them, because of the need for things to stay synchronous  -- one processor will usually be waiting on the other to complete it's task.

The what is taking so long on the official ARM version is that @chamnit just doesn't have the time to commit to the project that we would like. He's got a new kid, and a full time day job, and recently moved across the country.

If I was rich, I would hire him full time to work on Grbl, and hire a nanny too so that he can sleep at night. But, alas, I'm not; so the best thing I can do is try to contribute to the project as I can.

I know that @MeJasonT is aware of these links, but I'll leave them here for anybody else reading this thread.

https://github.com/gnea/grbl-Mega  --  The official Grbl on Mega 2560 version.
https://github.com/terjeio/grblHAL  --  A 3rd party ARM effort.
https://github.com/fra589/grbl-Mega-5X  --  A 3rd party 5-axis extension for Grbl Mega.
https://github.com/HuubBuis/grbl-L-Mega  -- A 3rd party lathe extension for Grbl Mega.
https://github.com/bdurbrow/grbl-Mega  --  My fork of Grbl Mega that adds support for 3D-printer derived UI hardware and some additional features.

MeJasonT, 2019-11-11:
I keep spiking my interest when Sonny mentions 6th order motion planning but alas it sounds like some of the upgrade issue are with the ARM Processors themselves.  

I'm currently looking at the Mega and Ramps combo for a possible 5 axis build, I already have a trunnion for 4th & 5th axis - its out of a paint mixing machine (the type the rotate and spin the life out of the can) believe it or not, this thing is seriously over engineered.

Im terrified i start building and spending dosh and a huge push happens to go down ARM route.
3.3v is really not our friend, the electronics is not only going to require more filtering but conversion to 3.3v.  Most of us at one time or another have had issues with signal noise on a huge 5v signal using a lower voltage is not going to make things easier.

Garage also needs an extension.

I suppose writing a gui that supports multiple USB devices within the one application isn't beyond the realms of possibility.

Having worked with 8086's, Z80's, PIC's and motorolla 68000's, i have to admit the humble ATMega328 is an absolute beast.  To think the Atmega 8 and 16 had 40 pins in a larger device but came to their end of life due to limited memory and resources - if only the ATMega328 had 40 pins and 30k more memory.

I keep seeing these touch screen controllers on Ebay from our friends in the Far East
at £200 + not a bad price but i'm hedging my bets with grbl as i'm sure these things wont be upgrade-able of get any firmware updates - That's a risk to far in my book.

I hadn't appreciated that Sonny was the only lead on the project, I did know about his child, job and move and would most definitely not want to be the one to P him off.   We are in the presence of a modern  day Einstein.  He is the only coder i have come across who has filled a processor to the max and it still works exceptionally well running a CNC Milling Machine.

looks like i will have to dust off my soldering iron and get ready to solder 0805 and 0603 SMD components not to mention a 100 pin ARM processor.  Hopefully before i get old and my eyesight fails - some time next week.

If i win the lottery i'm getting a Haas, sorry Sonny nothing personal.

langwadt, 2019-11-11:
5V doesn't make much of a difference in terms of noise the thresholds for 5V TTL are similar to 3.3V CMOS. About the only thing 5V had going for it was that it can sorta drive MOSFETs directly 

here's a port to a ~$10 ARM board that will work with uno shields and still have plenty pins left,
https://github.com/IRNAS/grbl_stm32

bdurbrow, 2019-11-11:
If you buy a clone, the upgrade to an ARM system won't be too bad. I don't know where you are, but for me here it's about the cost of a dinner for myself and my dad at the local 50's-themed diner.

FWIW, I've got a breakout board that I'm hoping to get the gerbers for up on github soon; it uses the almost the same pinout as a RAMPS + 2004 RepRap Discount LCD; and is designed to work on both 3.3v and 5v Arduino Mega boards (so, you could just drop in a SAMD21 based Mega board, swap a jumper, and have it work). The sticking point of this, however, is that SAMD21 based Mega... I may have to cook one up to go with the breakout board, if the Arduino project doesn't come out with one... fortunately the QFP version of that chip isn't too hard to solder by hand, if you've got a good iron & good magnification.

As for signal-to-noise ratio; what I did on my breakout board design was to immediately buffer the IO lines coming from the Arduino with some logic chips; so that the lower voltage signals didn't have far to travel. As to the real-world effects of that... I dunno; it's still in testing.

Re: import touch screen controllers.
If it's not open-source, I'm not interested. 😄  For the money, I'm thinking that you could set up something with a Raspberry PI + an Arduino running Grbl for less $$$, and know that you can upgrade or modify it yourself. (Personally, I find the communication & redraw latency of an external controller to be a pain - that's why I implemented my text-based UI - but, opinions do vary).

Re: Sonny. We don't always see eye-to-eye on _everything_, but he's definitely a smart cookie... and I will say that he does understand the motion system better than I do.

terjeio, 2019-11-11:
A [Re-ARM](https://www.panucatt.com/Re_ARM_for_RAMPS_p/ra1768.htm) board in combination with (IIRC a slightly modified) Ramps 1.6 is another possible option - for up to 5 axes. It is based on a NXP LPC1768. I have these boards in possession, will soon order a programmer/debugger for the Re-ARM... Ports for LPC1769 should work with minor/no modifications?

MeJasonT, 2019-11-12:
Thanks guys i will take a look at your suggestions

For you guys wanting to get Circuit boards made, take a a look at
https://jlcpcb.com/quote#/
I've started using them for prototyping now as the cost is cheaper than buying blank pcb and etchant with the added bonus of getting a silk screen, double sided board with through plated holes.
They will even make the solder templates.
The board quality has out surpassed my expectations, in fact they have done better than a UK manufacturer i've used at work.  For a laugh we put a board in an oven to 180 deg C for 500 hrs and it came out slightly discoloured as expected with zero de-lamination.
My only bug bear is UK customs, once your order gets over £12 it gets to be a royal pain in the arrr.
You get a 20% import tax + royal mail £12 handling charge,  umm gets silly.  So i just order 1 project board at a time now, and in some cases 10 boards work out cheaper than 5 (of one design).

The scary thing about some of the feedback to this message thread is the number of different and competing processors being used.  I'm sure you guys are aware that each manufacturer requires its own programming interface and programming lead.  The IC architecture is very different between manufacturers IC's as well.  So far we have ARM, NXP and ST microelectronics.  At least with the humble beginnings using the Atmega328 was pretty standardised.  I have the unfortunate position of being damn good at machine building but have only enough basic programming knowledge to be dangerous.  To switch from Atmel using the arduino IDE and format is going to be a steep learning curve for many, I'm Surprised that Sonny has kept control of GRBL for this long.  There is starting to be a little bit of a free for all thing going on in the search for a 32 bit future.

MeJasonT, 2019-11-12:
terjeio -I did come across this board. I think i found it through a suggested link from bdurbrow to your site above.  its now in my Ebay watch list (so i can find it again).

just out of curiosity how do you add links to people in threads, I must apologise for copy and pasting names but that's why.

X3msnake, 2019-11-12:
@Jason

You should look into platformio is much better than the arduino ide and
supports a lot of processor architectures

MeJasonT, 2019-11-12:
X3msnake Good shout, thanks for that

terjeio, 2019-11-12:
@MeJasonT - to link to people you just prefix the name with @ (when you do you can even select from a list of users participating in the thread).

You wrote:

> I'm sure you guys are aware that each manufacturer requires its own programming interface and programming lead.

This is not entirely true - most of the boards I am working with either has a bootloader or an on board programming/debugging interface. All but the Re-ARM board uses a USB lead for programming, The Re-ARM is programmed (by the end user) via transferring the (pre)compiled code on a SD card. For development I prefer boards with an on board debugger, or those that has the option of using an external debugger.  What is true is that the programming environments differ quite a bit though - I like those that are Eclipse based: TI, NXP and STM. ESP32 soon(?).

However, a end user should preferably (IMO) be able to use a precompiled binary - easier to achieve if moving most compile time options to $-settings.

TI-boards can even be programmed _and debugged_ from a [cloud based app](https://dev.ti.com/) - no need to install an IDE...

bdurbrow, 2019-11-12:
@MeJasonT ---
> I have not been able to find it on Ebay as a Mega board footprint.

Yeah, that's the problem. Atmel makes a version of the SAMD21 with an extra 16 IO lines; to fill in the extra 34 IO lines I'm figuring that I'm going to need to either drop another microcontroller on the board; or use some IO expander chips hooked up via i2c or SPI.

As far as I know, nobody makes something like this already.

MeJasonT, 2019-11-12:
@X3msnake Ive gone to PlatformIO site and opened Visual studio as instructed. opened Extension and update manager but nada, nothing kirputs. i have Vs2013, selected online and entered platformIO in search.  what am i missing.  Oh why me, i've had to learn all this CNC machining stuff and now i have to learn viz studio as well, last time i looked there were only 24hrs in a day.  To say your help would be appreciated would be an understatement.

X3msnake, 2019-11-12:
@Jason

i will start a new thread later tonight and tag you.

i will also add info on how to flash stm32 with arduino ide. might help you
and others

non 48h days is the caveat of OpenSource :)

bdurbrow, 2019-11-12:
Re: JLC
I've had good luck with them so far. I always check on pcbshopper.com before ordering; but for the past 6 months or so they've been coming out on top both for the price per board, and the shipping to California.

Re: processors.
FWIW, with the exception of the ESP32; all the 32-bit processors mentioned so far are ARM cores. The peripherals - timers, serial & usb ports, and interrupt controllers are quite different across manufacturers, however - and that's the real issue; as you need a whole different set of drivers for them.

Re: IDE.
The big advantage of the Arduino IDE is that it doesn't have a lot of features - so it's less intimidating for the novice to use to get a Grbl setup going. I don't write code in it however; for Grbl I use mostly BBEdit (I'm on a Mac) and occasionally Xcode. The only thing I use the Arduino IDE for is running the compiler & flashing the chip.