gnea/grbl #762: How to turn off laser when jogging

kch1027, 2019-11-21:
Hello. Please advise me how solve below my problem
I made laser machine myself with grbl 1.1h firmware.
Everything fine, good working, But when i move X axis or Y even Z, under laserGrbl or other laser program,  always lase  turn on.

typed $32=0 or M5 S0 on the command line before moving axis, but starting jogging the machine  always laser ON.
However, I can control Laser ON or OFF as M3, M5 command and controlled power by S0 -S1000  through command line. But start reading Gcode file, ignore M5 command, always laser on condition.
I don't know what's wrong or how to update firmware.
Please help me

hairu, 2019-11-24:
Is it possible that LaserGrbl is turning on the laser and not Grbl?  Have you tried using a sender such as UGS that's not specific to laser cutting?

arkypita, 2019-11-26:
Sound very strange. LaserGRBL does not send anything less, anything more than "$J=G91 Xnnn Ynnn Fnnn" when perform jogging, [as per specific](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging). In addition, the commands it sends are visible in the console.

Maybe your laser TTL/PWM on off control can work reverse? Standard is 5v on, 0v off.

kch1027, 2019-11-27:
Thanks you comments.
I suspect that laser control board is something wrong. because, if laser stopped on the position, sometime working M3 comment, and controlled power from 1-1000, and also stop the coming out laser by M5 comments (D11 0V) However if moving laser, always  power on ( out D11 around 4.6V)
But sometime, not waoking no answer any comments, only out D11 4.6V.
Anyway, I will be check again and tried again as well