gnea/grbl #763: GRBL ignoring "abort/reset" pin state

bajticzek, 2019-11-27:
Hello, I have physically wired abort switch to pin A0 of Arduino. Electrically it's working fine (no bouncing), all the necessary resistors and a cap is wired correctly. GRBL registers the closing of the switch correctly. In any state (jog, homing, cycle) it immediately stops and resets. That's fine. But after the reset it doesn't care about the pin's state. I am using industrial e-stop switch with locked closed position. So it stays closed after operating the switch. GRBL shouldn't be able to do anything until the switch is released, right? Or is this (logical) behavior not implemented? The machine can home and receive G-Code without an issue and it doesn't care about the switch until it's depressed and pressed again. Am I missing something?

terjeio, 2019-11-27:
@bajticzek  No you are not missing something, it is not implemented (IMO in the logical way).

bajticzek, 2019-11-27:
Dunno... If the e-stop is activated, you should not be able to move the machine at all. It shouldn't even communicate with outside world. But I guess the abort input is really just for hard reset of the machine, nothing else. I will implement this function myself then.

cri-s, 2019-11-27:
For E-Stop, and machine-safety requirement (EU)  you need a start and
a stop button
and it is required that after power loss the machine does not start.
To implement this you need a relay that have a start momentary button
that enable the relay
and further the relay must be autoholding.
Now e-Stop have double contacts, normally NO and NC.
NC is connected to that autoholding relay and interrupt the
autoholding function.
NO is connected to the stop interrupt.
The relay interrupt the Arduino after 12ms (normally 5ms for the relay
and the additional 7ms come from additional capacitator and diode that
is dimensioned accordly).
This is then HW implemented and uniquevocable dimonstrable that it
works always, SW is weak and as example jogging work always
indipendently of error state and the error state can be cancelled.
Further dimonstrating the that the SW works as intendet is not  easy
and this involves checking all source code for all function and
checking optimizing code (generated asm) ... .

BR
Chris

bajticzek, 2019-11-27:
My machine is wired in this way: I have three relays on the mains circuit. One of them switches 230V to all power supplies, another one is autoholding relay and the third is in series with the autoholding relay and switches off the entire machine via software. Effectively after pluging the main cord into wall outlet the machine does nothing. There is no power applied anywhere. After pushing momentary rocker switch, the autoholding relay activates, the main supply relay activates and machine "boots". I can activate the third relay via software, which breaks the autoholding function and switches the machine off gracefully after a linux machine powers down. If a power loss is to occur, the machine shuts down ungracefully and doesn't start again until you press that rocker switch again. My e-stop button is configured as two NO switches in parallel (I can choose up to three switches and select if I want NO or NC). I know NO is not the way to go and after some wiki reading I will invert the pin function on grbl Arduino and use two NC switches in series instead. But still I had a feeling when there is a dedicated "abort pin" input in the grbl software, the machine shouldn't do anything if this input is held in active state during grbl boot. If it is not the way it works, I will alter the source code to suit my needs. I already used the unused pin on Arduino to output current state of the GRBL - if it is not in IDLE, I cannot control the pneumatic workpiece clamping, it is in previous state and you cannot do anything with it until the machine returns back to IDLE. If you could point me in the right direction what to alter in source code, that would be great, if not, I will figure it out myself and you can close the issue then.

terjeio, 2019-11-27:
@bajticzek For my 32-bit port I have modified the main protocol loop to handle this in IMO a better way. Perhaps you can get some clues from that - from line 73 onwards in [protocol.c](https://github.com/terjeio/grblHAL/blob/master/GRBL/protocol.c) . There are some other configurable options too such as blocking operation if any limit switch is asserted or homing is not sucessfully completed.

abdset, 2019-11-29:
Just a trick: 
What about to use a safety door feature as e-stop? this one holds the machine until you release the e-stop. But will not reset the controler, you decide what is convinient for you.

doppelhub, 2020-01-10:
For estop, I recommend connecting to the 328p's reset pin.  This guarantees that the microcontroller is dead.  Note, however that you cannot update the 328p's firmware with an estop connected as is required by certification (i.e. low impedance logic high on NC, low impedance ground on NO, and reset connected to common).  If you don't care about UL/CE compliance, you can just leave the NC line floating (and let the onboard pullup keep reset high).  If you do need UL/CE, you'll need to add a diode in series with the connection to the reset.