gnea/grbl #764: Serial Port Error 11 : Data Check (Cyclic Redundancy Check)

arinurcahyo, 2019-11-27:
I'm using Woodpecker Chinese Board with GRBL 1.1 on Candle 1.28 Windows 10 platform
baudrate is 115200

After 6 month usage, it started to display serial port error, which is happen randomly, sometime after 2or 3 minutes or happen after hours.
I've tried to swap usb cable, using lower version Candle and other application such as UGS, and even reuse my old laptop which use windows XP but the problem persist .

I've googled that error code but found nothing

Did my board damaged? 

[image]

bdurbrow, 2019-11-28:
That's not an error generated by Grbl itself... it's an error coming from somewhere between the USB chip that is on the Woodpecker board to the USB port on your PC.

Given that you've tried a different PC entirely; I'm going to guess the USB chip on the woodpecker is starting to fail. Note that this does assume that nothing else has changed that would cause electromagnetic interference - if there's EMI in the environment, that could also account for the problem.

arinurcahyo, 2019-11-28:
I'm going to swap woodpecker with Keyes v4 CNC shield. Hope the problem solved.

Also there are a lots of welding equipment, bar cutting and bar bending machine around my workplace and they're shared a power source with the CNC and computer. And since they're already there before the CNC and the problem occurred recently, I'm didn't take EMI problem into account. I'm also going to look for it.

abdset, 2019-11-29:
I have seen this issue before:

- Your both usb cables are bad, buy new good quality one (Some cables have a transparent cover where you can see a ground shield in it all over the cable, buy these ones) 
- Or a USB connector on Woodpecker Board has a lost soldering on its pins.

bdurbrow, 2019-11-29:
Re: EMI. For what it's worth... a bad ground connection on a shield on something like a plasma cutter or welder (or just a motor) can cause EMI issues with another machine; without having the plasma cutter/welder/motor/etc showing any symptoms itself. If possible, relocation one or the other machine to increase the distance between them should show if one is interfering with the other.

Slixxor, 2020-02-25:
Sounds like an EMI issue. We had this issue when adding a high frequency plasma cutter to our machine. The USB drops out is the cause. Changed to a low frequency pilot arc plasma cutter and she's all good!