gnea/grbl #765: Laser on full power for 2-3 sec on every boot/connect.

AndKe, 2019-11-27:
version 1.1h : Everytime the nano is booting, or host connects, laser output is 100% for 2-3 seconds.
This is dangerous, as any reconnect or power issue that reboots the controller can enable laser at any moment.

bdurbrow, 2019-11-28:
At reset-time, all outputs of the ATMega328 chip go into "high-Z" mode - basically, they act like they aren't connected. To get rid of this issue, you need to connect a resistor to the laser enable input to pull it to the non-firing state (depending on your laser, that could be to 5v or to ground). Something like 100K ohms should be a reasonable value for this.

When Grbl want's to fire the laser, it will overpower the relatively weak pull being exerted by the resistor and force the input to what state it needs to be (5v or 0v).

AndKe, 2019-11-28:
Hi. you are right, I was too quick thinking this was a part of the boot process.
it is the sw-ld73 laser driver that has some internal pullup , not even 10k pull-down does eliminate the issue.

I will investigate further and maybe modify the sw-ld73 driver board.

langwadt, 2019-11-28:
try a much smaller pull down, it might be driven by a bootloader

AndKe, 2019-11-28:
I did not want to go too low on the pull-down resistor, and rather correct the cause.
So this was the solution to my problem:
https://madhacker.org/sw-ld73-laser-driver-modification/

BTW: the reason I thought that GRBL actually set that pin high on boot, was that the LED on the Develpower 2.0 board lit up on every boot, I did not realize that it was due to the crazy pullup of the laser board.

closed 2019-11-28 by AndKe