gnea/grbl #766: I got Hard limit pressed from D11

Maxfilo, 2019-11-28:
Hi All,

I have connected my CNC to Arduino and laser (to make it a laser engraver :) ). Anyway the problem I have is when I start the program it automatically reports Hard limit pressed and that's it. PWM is connected to D11 on Arduino and I have no idea what to do with it. Laser mode is enabled, PWM is enabled (I use LaserGRBL soft) and I still got this report. Of course machine stops and shows an error. When I disconnect D11 and start the program machines works but of course without laser - but movement is correct. Any ideas ? Hard limit switches are connected to D9 on Arduino and they work. I don't know why D11 reports as limit switch. Maybe somebody has this issue and can support me with that.

neilferreri, 2019-11-28:
It sounds like your laser is causing interference, triggering a false detection of your limit switch. Do you have any EMI filter on your switch inputs? Shieled cable?

Maxfilo, 2019-11-28:
> It sounds like your laser is causing interference, triggering a false detection of your limit switch. Do you have any EMI filter on your switch inputs? Shieled cable?

Thanks for response, yes my cable is shielded and I'm using this laser https://gocnc.de/shop/en/store/cnc-acessories/for-next3d-series/445/6-watt-laser-head-with-co2-supply-market-launch?c=64 , so it is not a standard cheap solution. And as I understand D11 should be an "output" not input so there should be no readings from laser ? Any ideas ?

abdset, 2019-11-29:
Check this out and follow all StuartB4 comments, they should fix your problem.

https://github.com/arkypita/LaserGRBL/issues/368