gnea/grbl #768: Bresenham's algorithm for arcs

VladimirLusin, 2019-12-03:
Hello. I'm from Russia. I use automatic translation from Russian into English, so I apologize if the text is not entirely correct.
I am making my simple Atmega CNC system. I got acquainted with your experience and am going to partially use your ideas, in particular, the ring buffer for receiving program frames from the computer and the ring buffer of the motion planner.
I wonder why you use approximation of arcs by line segments? I think it's hard. Calculation of trigonometric functions is required, and a lot of segments need to be calculated.
After all, there is Bresenham's algorithm for constructing arcs. It happens in integers.
There is also a Michener's algorithm (this is an improved Bresenham algorithm).
In industrial CNC racks, the evaluation function method is also used. It is even simpler than the Bresenham algorithm.
I also found a description of the CNC rack manufactured in the Soviet Union, where the method of a digital differential analyzer in integers was used. I already tried it. At first he simulated on a computer - he drew circles. Then used in the firmware for Atmega. My CNC plotter drew a perfect circle.
If you are interested in these algorithms, I could share my codes.

bdurbrow, 2019-12-03:
The motion engine (planner & stepper driver) requires line segments. They can be very small, but the format is line segments, not random-ordered points like a raster image buffer can handle. 

The classical arc algorithms are based on being able to plot points in random order - this is impossible with CNC; as you must follow the arc from beginning to end. So, you would have to re-execute the arc algorithm as it crossed each quadrant; this eliminates several of the the optimization opportunities the classical algorithms implement. On an ATMega328, there isn't enough RAM to hold the whole set of points at once; they are generated as the CNC traverses the arc.

It may be possible, even with these limitations, to improve on the performance of the arc generation algorithm; but timing testing with an oscilloscope would be needed to verify the results (turn off interrupts, toggle a pin high when you enter the calculation, and toggle it low again on exit, re-enable interrupts; and use the oscilloscope to see how long the pin was high).

Practically, at the moment the existing algorithm seems good enough; and all of Chamnit's available time is being spent on the ARM port of Grbl. This is not to say that you shouldn't experiment with it; this is just to say that I don't expect _him_ to do anything with it until the ARM version is out and well-settled.

😄

VladimirLusin, 2019-12-03:
I do not criticize the existing algorithm GRBL. I respect people who create something. Just thinking to suggest how to simplify the algorithm.
Many CNC systems perform circular interpolation using the DDA method or the valuation function method. That was already 30 years ago and even earlier. Atmega328's capabilities were fantastic at the time.
I consider that it is not necessary to put all points of an arc in the buffer. They can be calculated in real time. For the planner, you need to calculate the velocity vector at the entrance to the arc and at the exit from it. Most likely, speed can also be controlled in real time, having calculated in advance in which place you need to start accelerating or decelerating. I have not done this yet, but I will try.
Well, I will not bother with advice. Continue your work. Thank you for sharing your best practices. Good luck!

langwadt, 2019-12-03:
X and Y will be accelerating and decelerating throughout the arc

VladimirLusin, 2019-12-03:
Yes it is. With uniform movement around the circle, X and Y will move with changing acceleration.
I meant the total acceleration. After all, you can start moving along the arc at a low contour speed, if necessary, then, gradually increasing the speed, gain the necessary speed, and at the end of the arc you can also smoothly reduce the contour speed, if necessary. The total acceleration can be calculated as the sum of the vectors of tangent acceleration and centripetal acceleration (V * V / R).

chamnit, 2019-12-03:
The circular Bresenham was evaluated almost 8 years ago. No need to over complicate the planning and step algorithms to accomodate circular arcs. Keeping things simple is always more important. 

Also Grbl’s arc routine minimizes the trig computations needed by using a small angle coordinate transformation instead. It’s efficient enough to run on a slow 8-bit AVR.

closed 2019-12-03 by chamnit