gnea/grbl #769: JOG commands turn GRBL into "dumb" mode

luben111, 2019-12-03:
Hi,

I'm using 1.1h GRBL on Arduino Nano connected through opto isolated shield to the CNC. Everything works fine except that I can sometimes go into some "dumb" mode where the machine can't execute anymore some commands.
The scenario to go into this "dumb" mode is the following:

1. Connect to the GRBL, make homing cycle and position the head 5mm away from the origin - this work flawless. 

2. By using my PC software I'm sending jog commands to the GRBL to move the head left and right by 1 mm. 
\85$J=G91X-1F4000\n  and
\85$J=G91X1F4000\n  

3. There is background task  which is polling every 250ms the head position by sending ? (question mark)

4. If I keep pressing left and then right jog buttons quickly one after the other (once in a row)  and by keep pause between groups to allow head to settle down (and the input buffer to free) after some time I can see the following:
  a) GRBL doesn't accept jog commands - sending them will not move the head and doesn't return OK (no errors as well)
  b) GRBL doesn't execute also movement commands
  c) GRBL continues to output the status on reply to ?

Here is a snapshot of my debug output - it shows everything I'm sending and receiving on serial port (... is the character for 0x85):

o-  …$J=G91X-1F4000
-> ok
o-  …$J=G91X1F4000
-> ok
o-  …
o-  …$J=G91X-1F4000
-> ok
o-  …$J=G91X1F4000
-> ok
o-  …
o-  …$J=G91X1F4000
-> ok                                  <<<< this is the last time I see OK reporting, the problem occurred!
o-  …$J=G91X-1F4000
o-  …
o-  …$J=G91X-1F4000
o-  …$J=G91X1F4000
o-  …
o-  …$J=G91X1F4000
o-  …
o-  …$J=G91Y1F4000
o-  …

And here is what I get as a status - it keeps running correctly all the time:

o-  ?
-> <Idle|WPos:-127.667,-12.000,-1.000|FS:0,0|Ov:100,100,100>

or 

o-  ?
-> <Idle|WPos:-127.667,-12.000,-1.000|FS:0,0|WCO:-10.000,-10.000,-10.000>

My question is - have you seen such behaviour in GRBL? Any ideas how to resolve this will be welcomed.

Kind regards,
Luben

sadezohdi, 2020-01-19:
Hi. That's what I realized ,If your G-codes are large and you want to send them immediately, the Jog cancel command will not work properly. I used the formula that GRBL itself provided to fix this problem.

luben111, 2020-01-26:
Hi sadezohdi,
Can you share how you used "the formula that GRBL itself provided"? 
Notice that I don't overflow the GRBL buffer, it's almost empty.

sadezohdi, 2020-01-30:
Please read this page
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging