gnea/grbl #770: Relay starts with on mode

regpye, 2019-12-04:
I am using GRBL 1.1f
I have a relay module connected to D13 on an Arduino nano that I wish to use to switch on my plasma torch.  I have it functioning OK in the software and am using M3 to turn it on and M4 to turn it off.  The problem I have is that when I power up or connect GRBL the relay switches on and stays on until I send a M4 to turn it off.  This is very dangerous because the plasma switched on before it should.
Is there a way to stop this happening?

bdurbrow, 2019-12-04:
Two possibilities come to mind; first you can use the startup blocks feature to turn off the relay, this is described in this wiki page:

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#n---view-startup-blocks

The other thing you can do is wire in an "arming" switch (perhaps using another relay setup such that it automatically turns off when power is removed). Then, if the arming switch/relay has not been activated, the main relay won't be connected to the plasma cutter.

regpye, 2019-12-04:
Thanks for that info.  I feel that the first option is not going to be very useful because the relay is triggered before even starting up. Also the startup blocks will not be activated until after the homing sequence, so the only option will be to use a switch until all has settled down.
I wish there was another way because it is very messy to have a secondary switch to activate when it should be a simple system.
The relay I am using BTW is a relay module

regpye, 2019-12-04:
There may be another option. The problem seems to be that the relay module is low level triggered, so that means that it will be normally active when first powered up.  I have ordered some high/low level relays to try out. 
https://www.ebay.com.au/itm/Hot-5V-1-Channel-Optocoupler-Relay-Module-for-Arduino-H-L-Level-Triger-/232395396642

cri-s, 2019-12-04:
You maybe should add pullup or pulldown resistor to the output pin.

regpye, 2019-12-05:
> You maybe should add pullup or pulldown resistor to the output pin. regpye <notifications@github.com> schrieb am Mi., 4. Dez. 2019, 22:55:
> […](#)
Do you know how to do that, because I am not familiar with the process..

bdurbrow, 2019-12-06:
It's just a resistor of reasonably high value (50k to 100k ohms) that goes between the output pin of the arduino and either ground (for a pull-down) or the 5v supply pin (for a pull-up).

The idea is that the resistor will always be trying to pull the output pin to what the other end is connected to (ground or 5v), and at startup the Arduino won't be driving the pin at all, so the resistor is the only thing that provides any level to the pin. However, once the pin does get driven by the Arduino; it will overpower the resistor (that's why it's something in the range of 50k to 100k - enough to provide some pull, but not so little resistance as to be a short circuit).

However; the length of time between being off and having the spindle direction pin being driven by the Arduino is very short. The default direction is forwards; and this gets set in less than a second.

Can you modify and update the firmware on your board? If so, then you can look in `spindle_control.c`, and add this line:

`SPINDLE_DIRECTION_PORT &= ~(1<<SPINDLE_DIRECTION_BIT);`

It should go at the end of the `spindle_stop()` function; in between lines 115 and 116.

Doing that will turn off the spindle direction pin whenever the spindle is stopped (regardless of the direction that the rest of Grbl thinks it's supposed to be turning - but as this is a plasma cutter, that's actually what you want - there's no motor drive & actual spinning bits to get confused or thrashed by this).

regpye, 2019-12-06:
Thanks for the very detailed explanation. I have learnt a lot from that.
I will have to learn a bit more about modifying the firmware and have a go.
The GRBL 1.1h that I am using came as a hex file, so I didn't learn much at all about compiling or making any changes.  I have down loaded the source file, so I will try my hand at it later this week.

regpye, 2019-12-06:
bdurbrow 

I have followed your instructions as I thought, but have this error message when flashing  my nano.

Arduino: 1.8.10 (Windows 8.1), Board: "Arduino Nano, ATmega328P"

In file included from C:\Users\Reg\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Reg\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Reg\Documents\Arduino\libraries\grbl/nuts_bolts.h:56:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Users\Reg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 

In file included from C:\Users\Reg\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Reg\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Reg\Documents\Arduino\libraries\grbl/nuts_bolts.h:57:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Users\Reg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 

In file included from C:\Users\Reg\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Reg\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Reg\Documents\Arduino\libraries\grbl/nuts_bolts.h:61:0: warning: "bit" redefined

 #define bit(n) (1 << n)

 

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Users\Reg\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.2\cores\arduino/Arduino.h:124:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 

Sketch uses 29020 bytes (94%) of program storage space. Maximum is 30720 bytes.
Global variables use 1543 bytes (75%) of dynamic memory, leaving 505 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
An error occurred while uploading the sketch
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x0a
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x47
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x72
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x62
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x6c
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x31
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x2e
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x31

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bdurbrow, 2019-12-06:
OK, the warnings issued about redefining min() and max() can be ignored, it's poor style but not an actual problem to do that.

The low memory error is normal - in the '328 processor Grbl does, in fact, run really close to the limits of it's RAM.

This error, however, is a problem. It means that either the Arduino IDE is not talking to the board (wrong port selected in the IDE? driver issue? or just a temporary glitch that will go away with a reboot?) or the board is not responding (it didn't reset properly? missing bootloader? bad cable?).
`avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x0d` 

Depending on how you uploaded the previous Grbl .hex file; you may have overwritten the bootloader. If so, you'll need a second device (another Arduino will work, or something like an USBASP or Atmel-ICE programmer) to flash the bootloader back onto your board.

If you haven't seen it yet, this is the current official documentation for compiling this version of Grbl:
https://github.com/gnea/grbl/wiki/Compiling-Grbl

regpye, 2019-12-06:
bdurbrow
OK, thanks.  I don't really understand it much, so I will preserver with it. I checked
 the nano with UGCS and it seems to be the same as it was before, the header reads that it is version 1.1f and the new version should show 1.1h, so maybe it didn't do anything, except all the $ settings are now different, maybe default values?? 
I just flashed a second nano and got the same results.

regpye, 2019-12-06:
I have found this in grbl.h 

#define GRBL_VERSION "1.1h"
#define GRBL_VERSION_BUILD "20190825"

but when I compile and flash to the nano it shows 1.1f instead.
Previously the nano had Estlcam firmware loaded on it and I overwrote with GRBL.
Prior to that was GRBL 1.1f, so I am not sure what is happening here.

chamnit, 2019-12-06:
Delete old versions of Grbl in your Arduino library path. Re import Grbl as a library. Also disable aggressive caching in the Arduino preferences.

regpye, 2019-12-07:
chamnit
OK, I have deleted all versions of GRBL, I found one in the documents folder and that is what was being used.  Have managed to upload 1.1h now.  I couldn't find ant reference to  aggressive caching anywhere, I also checked the text file for preferences.  Not got the same errors as before, just the warning of low space which I believe is normal for a nano.

regpye, 2019-12-07:
After flashing GRBL 1.1h I still have the original problem with the relay module switching on at startup.
I am using UGCS to send gcodes to GRBL and am using M3 to turn on the relay module and M4 to turn it off.
On startup or resetting GRBL, the relay comes on and I have to wait until UGCS has fully loaded before I can press the button that sends the M4 to turn off the relay module. In real terms, this would mean that the plasma torch is firing for several seconds or until I am able to press that button.

My next action will be to try adding a pull up resistor, and if I understand correctly, this needs to be a resistor placed between the signal pin of the relay module and the 5v pin on the nano. Please correct me if I am wrong, I don't want to blow a nano for no reason.

I have added the line as suggested by bdurbrow 
"Can you modify and update the firmware on your board? If so, then you can look in spindle_control.c, and add this line:

SPINDLE_DIRECTION_PORT &= ~(1<<SPINDLE_DIRECTION_BIT);

It should go at the end of the spindle_stop() function; in between lines 115 and 116."

I don't see much difference or a change in the problem I have.

regpye, 2019-12-08:
If I use a timer relay module to activate the pin D13 to the relay module signal pin, I can overcome the problem, but only if I can send an M4 on start up (maybe)
It possible to send an M4 using a start block by $N0=M4
I tried this and as GRBL starts up, I only get a short flash of the relay being on and then it goes into the wanted off mode.  If I add a relay timer that has a few seconds delay, I can put that between the D13 pin and the relay module signal pin, that should work and fix the problem I think.
I have tested the starting block and it works, next I will wire in a delay relay and test that.
Seems to be over kill for a simple action, but no-one seems to know a way around this problem that works.

bdurbrow, 2019-12-08:
> I don't see much difference or a change in the problem I have.

Doh! I missed this earlier:

> The problem seems to be that the relay module is low level triggered, so that means that it will be normally active when first powered up.

That line I suggested to add to spindle_stop() sets the output pin to low, not high. Try this instead:

`SPINDLE_DIRECTION_PORT |= (1<<SPINDLE_DIRECTION_BIT);`

That will set the output high.

regpye, 2019-12-08:
bdurbrow 

I have changed the line to SPINDLE_DIRECTION_PORT |= (1<<SPINDLE_DIRECTION_BIT); and it made no difference to the problem.
I have added the start block in GRBL to send an M4 on startup, and have tested using a delay relay to connect the line between D13 and the relay module, and it is working that way.  A bit of mess due to having now three boards, two of which are to just turn on a switch. But it works.

regpye, 2019-12-12:
Today the high/low level relay modules came, so I set them at the high level and wired them up again. The problem was from using low level type relay modules.
All is working fine now. I was also able to get M5 working to turn off the plasma torch module and M4 to turn it on.  The coolant is also working with M8 on, and M9 off.
I would like to thank all the members that tried to help me, it is very much appreciated.

regpye, 2019-12-22:
Thanks for all the help and suggestions guys.

closed 2019-12-22 by regpye