gnea/grbl #771: Issue with arcs

SketchThis, 2019-12-06:
Hello everyone, I'm having an issue with arcs in GRBL and I'm trying to find out if it's my CAM software's fault, or GRBL...

The CAM software I'm using is outputting 4 arcs for circles. We're testing this on 2 GCode based machines and it works on one machine, but not our GRBL machine. The arcs are getting cut off, it's hard to explain but you can see it in this video:

https://photos.app.goo.gl/UCbk9SuZEWvYvNkN6

I've attached the G Code we're using, along with the errors in the GRBL console.

What's odd is that in the G Code previewers we're using it looks fine, and on our Centroid machine it runs nice and smooth circles. According to the GRBL wiki these arc commands should be supported, and as far as I can tell they're in there correctly.

I'm struggling to identify the source of the problem, is it the CAM that generated this file, or GRBL? Any help would be greatly appreciated!

[test 1.nc.zip](https://github.com/gnea/grbl/files/3933663/test.1.nc.zip)

[errors.txt](https://github.com/gnea/grbl/files/3933668/errors.txt)

terjeio, 2019-12-07:
From the linuxcnc docs:

"When programming arcs an error due to rounding can result from using a precision of less than 4 decimal places (0.0000) for inch and less than 3 decimal places (0.000) for millimeters."

SketchThis, 2019-12-07:
AHHHHHH, I'll test that!

SketchThis, 2019-12-09:
Can confirm it works!! Thank you @terjeio

closed 2019-12-09 by SketchThis