gnea/grbl #772: Homing doesn't work but Hard Stops do

taaron17, 2019-12-11:
I'm trying to figure out how to get my homing sequence to work but the servos don't seem to move. 

The board I'm using is a makerbase mks dlc v2.0. I have a single X axis motor and two Y axis. There are no Z axis motors being used. It's worked fine with LaserGRBL as far as just using it out of the box. The endstops I'm using are these (https://www.amazon.com/gp/product/B06XTB7WMK/). They plug directly into the endstop and the board. When I power on the board and press an endstop, the LED light on the end stop turns on. 

I enabled the homing sequence, and upon pressing the home button, no servos move. Eventually it will just fault out.  I have tried with both $5 = 0 and 1.  I enabled the hardstops and jogged the system. When the laser hit the endstops it stopped the jog sequence immediately. So it's recognizing the end stops as endstops, but is having issues going into the homing sequence. Has anyone run into this before?

langwadt, 2019-12-11:
did you change the homing sequence to only do  X and Y? line 105-113 in config.h

taaron17, 2019-12-11:
I've just been using the installable version of the machine. Is there a way to do this without having to compile the code yourself?

bdurbrow, 2019-12-12:
No, you're going to have to re-compile Grbl to have configuration options from the .h files (config.h, cpu_map.h) take effect.

This wiki page has info on how to do that.
https://github.com/gnea/grbl/wiki/Compiling-Grbl

night022, 2020-08-21:
Hello, do you have a solution for your problem?