gnea/grbl #773: Homing does not move CNC maching Alarm 9

BrendonWright, 2019-12-12:
Hey i have a BobsCNC and am trying to figure out why when i HOME it does not move. Wires have been checked everything seems to be hooked up properly per the manual. Using UGS platform. GRBL 1.1 with an arduino uno card. Here are my firmware settings. 

[Error] An error was detected while sending '$H': (ALARM:9) Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring. Streaming has been paused.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 1.1h ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 1    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 250.000    (Homing locate feed rate, mm/min)
$25 = 4000.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 80.000    (X-axis travel resolution, step/mm)
$101 = 80.000    (Y-axis travel resolution, step/mm)
$102 = 2267.717    (Z-axis travel resolution, step/mm)
$110 = 10000.000    (X-axis maximum rate, mm/min)
$111 = 10000.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 500.000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$122 = 300.000    (Z-axis acceleration, mm/sec^2)
$130 = 610.000    (X-axis maximum travel, millimeters)
$131 = 610.000    (Y-axis maximum travel, millimeters)
$132 = 85.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0.0]
ok

ONE thing i did notice was. when i push the z axis home switch/limit switch. is reads as X and Z This cannot be normal. Would it be a card issue? or  plugged into the wrong pins? I have checked and recheck all the wires everything is set and hooked up like the manual says its suppose to be hooked up. Does anyone have any ideas?????

BrendonWright, 2019-12-12:
Re flashed and the Z home switch is fine now

BrendonWright, 2019-12-12:
[Error] An error was detected while sending '$H': (ALARM:9) Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring. Streaming has been paused.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 1.1g ['$' for help]
>>> $$
$0 = 5    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 1    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 500.000    (Homing locate feed rate, mm/min)
$25 = 8000.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 80.000    (X-axis travel resolution, step/mm)
$101 = 80.000    (Y-axis travel resolution, step/mm)
$102 = 2267.717    (Z-axis travel resolution, step/mm)
$110 = 10000.000    (X-axis maximum rate, mm/min)
$111 = 10000.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 800.000    (X-axis acceleration, mm/sec^2)
$121 = 800.000    (Y-axis acceleration, mm/sec^2)
$122 = 300.000    (Z-axis acceleration, mm/sec^2)
$130 = 610.000    (X-axis maximum travel, millimeters)
$131 = 610.000    (Y-axis maximum travel, millimeters)
$132 = 85.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0.0]
ok

FelisucoVFR, 2019-12-12:
You have to enable $21 = 1 to use the limit switches.

BrendonWright, 2019-12-12:
[Error] An error was detected while sending '$H': (ALARM:9) Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring. Streaming has been paused.
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 1.1g ['$' for help]
>>> $$
$0 = 5    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 1    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 500.000    (Homing locate feed rate, mm/min)
$25 = 8000.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 80.000    (X-axis travel resolution, step/mm)
$101 = 80.000    (Y-axis travel resolution, step/mm)
$102 = 2267.717    (Z-axis travel resolution, step/mm)
$110 = 10000.000    (X-axis maximum rate, mm/min)
$111 = 10000.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 800.000    (X-axis acceleration, mm/sec^2)
$121 = 800.000    (Y-axis acceleration, mm/sec^2)
$122 = 300.000    (Z-axis acceleration, mm/sec^2)
$130 = 610.000    (X-axis maximum travel, millimeters)
$131 = 610.000    (Y-axis maximum travel, millimeters)
$132 = 85.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0.0]
ok

BrendonWright, 2019-12-12:
Still no movement :(

BrendonWright, 2019-12-12:
If i unlock and its IDLE shouldn't the machine move? If i use the Jog Controller? Because it does not and gives me an error

FelisucoVFR, 2019-12-12:
Try disabling the soft limits from now. $20=0

BrendonWright, 2019-12-12:
soft limits disabled and nothing

FelisucoVFR, 2019-12-12:
🤔$110 and $111 to 500 now. You can increase them later.

FelisucoVFR, 2019-12-12:
Any picture of the hardware?
If previous doesn’t work, it should be hardware issue.
Any noise?

FelisucoVFR, 2019-12-12:
Silly question: motors power supply is on? Checked with multimeter?

BrendonWright, 2019-12-12:
Okay did the 110 and 111 to 500 still nothing 

Picture of the adruino card? 

Motors have power they are locked up and ready to move they just arn't moving. The fuse on the card is good

chamnit, 2019-12-12:
I would check if it moves at all with regular gcode. BobCNC controller boards jump the stepper driver enable pin physically to keep them enabled all of the time. It can give the false impression that Grbl is properly connected or configured.

FelisucoVFR, 2019-12-12:
@BrendonWright agreed with @chamnit.
With that config, it should move, correctly or not, so I would point to grbl pinout config or other hardware issue.
Which hardware are you using?

FelisucoVFR, 2019-12-12:
You can try to toggle the following one by one. These depends on the hardware.
Specially all my CNC's ar $5=0
$2 = 0 (Step pulse invert, mask)
$3 = 0 (Step direction invert, mask)
$4 = 0 (Invert step enable pin, boolean)
$5 = 1 (Invert limit pins, boolean)

FelisucoVFR, 2019-12-12:
Did you check the web at [https://support.bobscnc.com/hc/en-us/articles/360009104434-Diagnosing-Home-Switches-My-CNC-will-not-Home-Correctly-](url) ?

BrendonWright, 2019-12-12:
yes i checked that out all my home switches work properly and show up on UGS when engaged. I think what i have found is that my motors might not have the right voltage. the controller reads .63 and .64 volts for each stepper motor. Bobs page on this says its suppose to read .72... also my X1 motor is humming and has a slight vibration when i hold the motor. none of the other motors do that. all the motors are warm except my Z motor as well. Z motor is COLD. like its not getting any juice. but it reads .63

BrendonWright, 2019-12-12:
using Arduino/Uno FelisucoVFR

FelisucoVFR, 2019-12-13:
Finishing my ideas...
Don't assume warm is good and cold not. Mine are normally cold unless lot of work. But one cold and other warm is good point to follow. Check jumpers configuration if they are. Some drivers use them and other not.

doppelhub, 2020-01-10:
I'm not familiar with Bob's CNC, but I've seen issues where grbl returns alarm:9 if the maximum step rate is too high exceeds 30E3*60.  You might try uncommenting MAX_STEP_RATE_HZ in config.h, then try to write $110 = 10000 (as it is now).  If your parameters are problematic, grbl will now tell you so.

Beardy241, 2020-09-07:
I am getting the same problem: Alarm 9 on homing.

Did you ever solve this problem?

I've had the device running v0.9 for ages and it worked a charm.  Now trying to drive it from Lightburn I've re-flashed with v1.1h and it won't home.  I can move the axis so I know it's working.

Beardy241, 2020-09-08:
Yay!  I get to solve my own problem… which may help someone else!

As my device is a home-brew laser engraver, I eventually came to realise that if the 'CNC' thinks it has a Z axis and is (commonly) trying to home that first… it'll never get there — hence the ALARM:9

A quick gurgle search turned up **this page which talks about lasers and disabling Z homing in the config.h file** of the Arduino code.  https://github.com/gnea/grbl/issues/278

After running through the whole setup once again it's working nicely.

NB.  when reflashing the GRBL firmware, remember to 'fix' the config.h file to only home 2 axis!!

jenny00gray, 2020-10-13:
Did you ever get it figured out? I am having the same problem

Beardy241, 2020-10-13:
I came to the realisation that it was waiting for the Z axis to home (which it does first) but of course the laser has no Z so it waits and times-out. 

You have to poke in the grbl firmware; there are lines to comment/uncomment which tell it not to home Z. 

Works a charm now.