gnea/grbl #774: Displays the running gcode

sadezohdi, 2019-12-14:
Hi
I want to control GRBL with another Ardino. Almost eighty percent of the work is completed.
I send the G code and get the machine position correctly, but I miss the line that is running!
At first I was doing the character 'd' (idle) but I found it to be wrong. What is your idea of this?
Thanks for your great code ...

PicEngraver, 2019-12-17:
Try the character '?' to get position data???

"**Status Reporting**
When a ? character is sent to Grbl (no additional line feed or carriage return character required), it will immediately respond with something
like <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000> to report its state and current position. The ? is always picked-off and removed 
from the serial receive buffer whenever Grbl detects one. So, these can be sent at any time. Also, to make it a little easier for GUIs to pick up 
on status reports, they are always encased by <> chevrons.

Developers can use this data to provide an on-screen position digital-read-out (DRO) for the user and/or to show the user a 3D position 
in a virtual workspace. We recommend querying Grbl for a ? real-time status report at no more than 5Hz. 10Hz may be possible, but at 
some point, there are diminishing returns and you are taxing Grbl's CPU more by asking it to generate and send a lot of position data."

https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl

bdurbrow, 2019-12-18:
At least on Mega, there's an option to return the N number of the current planner block; I think it's on the '328 version as well. In my code base, it's normally on; and controlled by the `REPORT_FIELD_LINE_NUMBERS` config directive. You might do a search in the code base for `REPORT_FIELD_LINE_NUMBERS` to see if that will do what you want.

sadezohdi, 2020-01-19:
Thanks for your answer .
@bdurbrow 
This line is enabled in config.h. But when receiving the status report ('?'), it Does not send line number .

LiorBird, 2021-01-11:
> Thanks for your answer .
> @bdurbrow
> This line is enabled in config.h. But when receiving the status report ('?'), it Does not send line number .

Hi there, did you ever figure it out? I'm running into the same issue and can't really find anything online. Any luck?

GrahamJB, 2021-10-15:
My guess is that it is normal (and it is a guess) Line numbers basically aren't used in Grbl and they are not needed in the Gcode. The purpose of line numbers was originally to support things like subroutines, macros...... and these are not supprted by Grbl.

Also consider the processes that are gone through. The input is parsed and placed in the planner buffer, the operations are taken out of the planner buffer and turned into movement commands. The ? reports as accurately as it can the current position but the line number has long since been discarded, new lines may have already been parsed and added to the planner buffer.

I am interested to know why is the line number corresponding to the exact position important to you.

ecncshopInd, 2021-10-17:
I have added a logic in my Front end ; to ensure the DRO vals (X1,Y1) has not deviated from the current executing gcode line. Say Xstart, Ystart and Xend Yend are the start and end points of the currrent gcode. So iam checking if X1 Y1 is always lying on the line joining start and end point. This logic is failing sometimes, bcos of the line number mismatch.

ecncshopInd, 2021-10-17:
One more scenario: I have a Zdown move , then a small geometry and then a Zup move. 
I need to know reliably whether Iam in Zdown or Zup move , as I need to take some action (Mcode) only after Zup is completed. So I dont dump the Gcode after Zup , till I know for sure Zup is completed etc. I can think of few other scenarios, where reliably knowing current executing line number will be useful.