gnea/grbl #775: Version statistics

arkypita, 2019-12-16:
Hi @chamnit 
This is the usage statistics collected from LaserGRBL telemetry.
Based on 75000 unique installation this is the distribution between different grbl versions:

Version | Count | Percentage
-- | -- | --
1.1f | 34485 | 45,63%
0.9j | 15013 | 19,87%
0.9i | 9177 | 12,14%
1.1g | 4952 | 6,55%
1.1h | 3598 | 4,76%
1.1e | 2702 | 3,58%
0.8c | 2512 | 3,32%
1.1z | 1226 | 1,62%
0.9h | 984 | 1,30%
0.9g | 563 | 0,74%
1.0b | 243 | 0,32%
1.0c | 49 | 0,06%
1.1l | 22 | 0,03%
1.1i | 13 | 0,02%
0.9 | 10 | 0,01%
others | 25 | 0,03%

[image]

arkypita, 2019-12-16:
Of course this is not an issue/problem. I shared this data with you just only FYI and statistic purpose.

As LaserGRBL is specific tool for laser engraving it is very very normal that people is moved to upgrade to latest 1.1 version that has specific optimization for laser. In the real world that include a lot of CNC machine i cannot guess what could be the ratio between old 0.9 and new 1.1

MeJasonT, 2019-12-17:
Its interesting to see the growth of GRBL, what your data shows very clearly is the expansion of interest in home built CNC of any flavour.  Are the time periods between version upgrades similar. Or has V1.1f been out for a considerable length of time now and not upgraded any further as a result of researchers/coders looking for the leap to 32 bit processing.

arkypita, 2019-12-17:
According to the information I gathered from the pages of grbl these are indicatively the release dates of the various official versions.

Version | Release date
-- | --
1.1h | 25/08/2019
1.1f | 01/02/2017
1.1e | 10/12/2016
0.9j | 17/03/2016
0.9i | 29/03/2015
0.8c | 01/04/2013

arkypita, 2019-12-17:
> Its interesting to see the growth of GRBL, what your data shows very clearly is the expansion of interest in home built CNC of any flavour

I also share your impression. Looks that the number of people who buy a CNC, an engraver, or a 3D printer is continuously increasing.

This is also demonstrated by the ever increasing number of sites, communities, facebook groups, and even by the increasingly high number of "ready to assemble" machinery vendors from china.

I think that the simplicity and cheapness of grbl based systems has given a big boost to this phenomenon starting from its first release.

bdurbrow, 2019-12-18:
I also think the whole 3D printer boom fed into this; of the people who got one and stuck with it long enough to iron out the kinks (bed adhesion, learning 3D CAD, etc), they then went "Hmm, I wonder what _else_ I can do?"

chamnit, 2019-12-18:
Thanks @arkypita ! Two things strike me as interesting. There are lots of people using v0.9, even though v1.1 is way better at laser jobs. The other is I have no clue what Grbl v1.1z is.

MeJasonT, 2019-12-18:
If only I had this idea for data collection when I was studying statistics for A level maths

wmarkow, 2019-12-18:
>The other is I have no clue what Grbl v1.1z is.

Someone forked your repo, made some changes and released under **1.1z**?

arkypita, 2019-12-19:
> Thanks @arkypita ! Two things strike me as interesting. There are lots of people using v0.9, even though v1.1 is way better at laser jobs. The other is I have no clue what Grbl v1.1z is.

I think that only few people are really DIY enthusiast that buy spare parts, project and assemble their hw  and flash grbl from source (or precompiled binary). The most of grbl users are people that buy CNC or laser kit where all came preconfigured and installed. A lot of that kit come with v0.9

From the level of questions I receive looks that most people doesn't ever know the role of grbl in their hw and the difference between grbl and LaserGRBL (or any other UI).

Those people never update their firmware unless someone tell them to do, and how.

doppelhub, 2020-01-10:
I'll throw in my data: we've got 10,000+ units in the field:
1.0c: 8000 units
0.9g: 10000 units

I'm in the process of updating our machine firmware to grbl 1.1g... our custom "GrBLDC" PCB has quite a few additional features (stepper power levels, auto-level, realtime feedrate reduction when spindle is overloaded, etc), and getting them to all play nice has been a challenge with the many modifications from 1.0c to 1.1g.  I'll get there.  grbl is amazing as always.