gnea/grbl #777: Hi experts

samehherit, 2019-12-24:
Hi ,, 
I have a cnc home made , im using grbl 1.1
, candle software
, im using a power supply 12 volt 250 watt
And the spindle is apt grinder 600 watt
, when im running the file , the machine suddenly stoped working and the candle shows un active and freez
This message appears ,, see the foto belw 

But when im trying to simulate by running 
the file without turning the spindle on 
Th machine running normally without any stoping .  
 ,

SailWithChips, 2019-12-24:
Most provably noise problem. Try to separate steppers cables and spindle power cable, shield all of them and corretly gnd shields... provably with these easy steps will solve your problem.

markcomp77, 2019-12-24:
I agree with the previous speaker
- you must reduce the production of interference / noise
- separate the input cables (step / dir) and outputs (stepper motors and spindles) in space

but I will add that
it is worth improving the noise immunity

Arduino and various usb cables are highly susceptible to interference.
A better thick USB cable will help.
Changing the design of the Arduino module, introducing USB isolation, additional Step / Dir buffers, input optoisolation - this will also improve resistance to interference, suspension of transmission or the entire processor ...
Noise immunity is difficult to achieve without good grounding (low impedance) ... screens connected only to good grounding help.

samehherit, 2019-12-24:
> Most provably noise problem. Try to separate steppers cables and spindle power cable, shield all of them and corretly gnd shields... provably with these easy steps will solve your problem.

Thanx mr sailwithchips for the valuable information 💐💐 its really a logic reason 
, But let me ask about the material i can depend on to shield the stepper and spindle cables ? 

And how to gnd shields ? Because im a beginner

samehherit, 2019-12-25:
> I agree with the previous speaker
>
> * you must reduce the production of interference / noise
> * separate the input cables (step / dir) and outputs (stepper motors and spindles) in space
>
> but I will [...]

Thanx mr markcomp77 💐💐
About your great informations  
But i would like to know how i can achieve agod grounding ?

markcomp77, 2019-12-25:
As a good grounding I understand the property of the electrical installation to which we attach the machine (PE - usually yellow-green).
The building's ground connection goes to the ground. What grounding impedance we get depends on the properties of the earth in which we put the grounding electrode ... on the size of the grounding electrode ... on the depth of burying the grounding rod ... The best and recommended grounding for the building is the steel reinforcement of the foundation footing - very durable, because it is not subject to corrosion .. .

When we build a larger machine,
especially when cutting with plasma (inverter) and
spindle inverters> 2kW
grounding improvement involves burying your own grounding electrode. But to make it safe and in accordance with the regulations - it is worth asking the electrician to connect such grounding to the rest of the installation (to the leveling bar).

samehherit, 2019-12-25:
> As a good grounding I understand the property of the electrical installation to which we attach the machine (PE - usually yellow-green).
> The building's ground connection goes to the ground. What [...]

Thanx for usefully information

samehherit, 2019-12-30:
After modification based on your advices ,, the cnc is working nice without freezing 
,, Thank you very much