gnea/grbl #779: 有限位开关并且正常工作，每次连接连接电脑时如何找回之前的工件原点？

yangyi-sz, 2019-12-28:
目前的grbl每次连接电脑都会重置坐标信息，如果断开电脑和电源，重新连接，我该如何找回之前的工件坐标呢？

sadezohdi, 2020-01-18:
yes yes! :))
Please write english  ...

MicroTransactionsMatterToo, 2020-02-02:
@yangyi-sz Going by the translation from Google, regarding workpiece coordinates, they're saved to the EEPROM, so they remain when you start the machine back up. Since they're relative to the machine coordinates, you'll need to home the machine for the workpiece coordinates to be correct after a hard reset. 

谷歌翻译的歉意，希望对您有所帮助。

根据来自Google的翻译，关于工件坐标，它们被保存到EEPROM中，因此当您启动机器备份时它们将保留下来。由于它们是相对于机器坐标的，因此您需要将机器放回原处，以便在硬重置后使工件坐标正确。

ningzhang-github, 2020-05-25:
在Github这样的多语言协作环境问问题，咱们最好翻译成英语来获得更多帮助，即使是机翻也行。
我觉得最简单但是成本最高的方式就是各轴安装高精度限位开关，每次开机Home一下就能得到机械原点，参考这个坐标系就能定位工件。

Eng version: In a multi-language cooperation platform like this, we'd better speak in English, even using a machine translation.
IMO, the simplest but costly way to do that is using high precision proximity switch as limit switches of all axis, then after a "home" action you can get exact machine coordinates. Refer to this coordinates you'll be able to locate your part on the table.

yangyi-sz, 2020-09-16:
看到你们的解答，我十分荣幸，grbl限位开关都正常连接工作正常，在开始雕刻时已经归零，开始雕刻时清空了xyz工件坐标，按照正常流程中途断电再重新连接电源时，只需要执行一次回归机械原点grbl系统就会自动计算出断电以前的工件坐标，但是grbl没有，我使用蜡烛来控制grbl，执行机械原点后，计算出来的坐标是有问题的，不是理想的坐标信息。
I am honored to see your answers. The grbl limit switches are connected and working normally. They have been reset to zero when starting to engrave. When starting to engrave, the xyz workpiece coordinates are cleared. When the power is cut off and reconnected according to the normal process, only You need to perform a return to the mechanical origin. The grbl system will automatically calculate the coordinates of the workpiece before the power failure, but grbl does not. I use candles to control grbl. After the mechanical origin is executed, the calculated coordinates are problematic and not ideal coordinate information .

yangyi-sz, 2020-09-16:
设备限位开关都是正常工作的，执行过机械原点探测，得到工件坐标和机械坐标都是0的状态，软件显示如图片0003，然后我移动xyz到50的位置，并且清空工件坐标，得到的图片0001的位置参数，如果此时断开电源再次连接电源时所有坐标都变成了0，得到图片002，即使我执行机械原点探测，系统也不会计算出断电以前工件坐标的位置，断电后重新上电，执行机械原点探测又会回到0003图片的坐标，这个问题苦恼了很久，希望有高人指点。（按照大家回复的信息，应该在断电后重新上电，执行机械探测应该得到图片0004的坐标参数，但grbl不是）
The limit switches of the equipment are all working normally. After performing the mechanical origin detection, the workpiece coordinate and the mechanical coordinate are both 0. The software displays as picture 0003. Then I move the xyz to the position of 50 and clear the workpiece coordinates. The position parameter of picture 0001, if all the coordinates become 0 when the power is disconnected and connected again at this time, picture 002 is obtained. Even if I perform mechanical origin detection, the system will not calculate the position of the workpiece coordinates before the power is off. After re-powering, the machine origin detection will return to the coordinates of the 0003 picture. This problem has been troubled for a long time, and I hope someone can give me advice. (According to the information you reply, you should re-power on after power off. Performing mechanical detection should get the coordinate parameters of picture 0004, but grbl is not)
[image]
[image]
[image]
[image]

yangyi-sz, 2020-09-16:
使用蜡烛来控制grbl，软件上面的复位按钮，也会重置坐标数据，会不会是因为grbl通电后，软件就自动复位了grbl才导致的坐标数据重置呢，有这方面的参数修改方法吗？
Use candles to control grbl, the reset button on the software will also reset the coordinate data, is it because the software automatically resets the grbl after the grbl is powered on, and the coordinate data is reset? There are parameter modification methods in this regard ?