gnea/grbl #781: Rotary engrave

moonface1977, 2019-12-28:
I have been using 1.1 for a while now and I thank the community for the support. I'm trying to engrave with a very pointy engraving bit, almost like with a laser. But the laser command doesn't control the up/down when no engraving. Is there a way to accomplish this.

beckmx, 2019-12-28:
you need a leveler for that

moonface1977, 2019-12-28:
Sorry, but I'm not sure what are you referring to. Level as in auto-leveing the z axis? Don't understand how that can help with this.

SailWithChips, 2019-12-28:
As far as I understand you are not using cooland on your configuration... I would suggest to use M8 before each engraving process and use this ‘M8 Output’ for an ‘up/down’ positioner, for instance a bistable or whatever you are using... with an M9 for switching that output off then return back to ‘up’ position... hope this idea could be helpful to you. Normally on the engraving programs you can set some macro whenever operation is fired so... possibly you can do it other way. On the other hand, if you provide complete information of your machine like for instance, you are using z-axis or not, would be easier for a better response. Wich engraving tootlpsth program are you using?

moonface1977, 2019-12-28:
My machine is running grbl 1.1 on a Arduino shield. Usually I use UGS, but I was trying with Lasergrbl

moonface1977, 2019-12-29:
What in trying to accomplish is for the bit to go up when not engraving (laser off) and down when engraving (laser on)

SailWithChips, 2019-12-29:
Yeah the idea is clear!, but what about your machine HW configuration?, GUI and gcode program generator?

moonface1977, 2019-12-29:
Sorry. Arduino UNO + cnc shield. Double Y motor. Fusion 360 ( I also tried with Laser grbl). UGS or rabbit for sender.
I hope this helps. Thanks a lot!

MARIOBASZ, 2020-01-01:
I do not know if I understood the question well, if what you need is for the machine to compensate for flatness errors (such as for milling the tracks on a pcb), you must do an autolevel, I use bCNC instead of the popular UGS, there are many and for me useful accessories (and active community), including autolevel.