gnea/grbl #782: Wrong text in Release Notes 1.1h?

ArjanDeVries, 2019-12-29:
Hi,

After reading the release notes of 1.1h im wondering if the text below is ok and if text "Y-axis" is wrong and needs to be X-axis. Also i have NC switches so connecting the other switch to the other z pin will not work. Setting this up in serial will be ok i think. Maybe this is also something to mention. TY for all your hard work!

Dual axis homing will automatically abort homing if one limit switch triggers and travels more than 5% (default) of the non-dual axis max travel setting. For example, if the X-axis has dual motors and one X-axis triggers during homing, Grbl will abort 5% of the Y-axis max travel and the other X-axis limit fails to trigger. This will help keep any misconfigurations or failed limit switches from damaging the machine, but not completely eliminate this risk. Please take all precautions and test thouroughly before using this.

chamnit, 2019-12-31:
No it's correct. 5% of Y-axis travel because you are trying to keep the other axis from racking too much if the X-axis motors aren't aligned.

closed 2019-12-31 by chamnit

ArjanDeVries, 2020-01-03:
Ah, i think i understand now. More x width of the gantry in my case make more y movement possible.