gnea/grbl #783: Dual axis problems

ArjanDeVries, 2019-12-29:
Hi!

I'm currently working to get the dual axis configuration working. But it's not yet working as expected.
My config:
I have an UNO with CNC shield that was working fine with the 1.1g version.
I have 2 motors on the y axis.
All endstops are NC and the second motor on the y axis (Motor A on the shield) has also an end-stop in series with the z axis.
When i trigger the endstop on Motor Y the status shows Y
When i trigger the endstop on Motor A the status shows YZ

It looks like the end-stop on Motor Y is ignored. Because on homing it stays moving when the end stop is triggered.

Does someone has any idea?

Looking at the code in limits.c this looks strange to me, because why take the y axis when the x axis is configured?
`#if (DUAL_AXIS_SELECT == X_AXIS)
      float fail_distance = (-DUAL_AXIS_HOMING_FAIL_AXIS_LENGTH_PERCENT/100.0)*settings.max_travel[Y_AXIS];
    #else
      float fail_distance = (-DUAL_AXIS_HOMING_FAIL_AXIS_LENGTH_PERCENT/100.0)*settings.max_travel[X_AXIS];
    #endif`

ArjanDeVries, 2019-12-29:
I measured the distance from the gantry to the end of the table before homing. And it seems that rotating the axis manual homing does not correct the change:

<Idle|MPos:-1.000,-1.000,-1.000|F:0|Pn:RHS> 41.5 41.5 (current distance left and right side 41.5)
<Start Homing>
ok
<Idle|MPos:-1.000,-1.000,-1.000|F:0|Pn:RHS> 41.5 41.5 (no problem, distance still the same)
Manual move of one axis
<Idle|MPos:-1.000,-1.000,-1.000|F:0|Pn:RHS> 41.8 41.5 (manual changed the left side 0.3 mm)
<Start Homing>
ok
<Idle|MPos:-1.000,-1.000,-1.000|F:0|Pn:RHS> 41.4 41.1 (the initial value 41.5 is changed???)

ArjanDeVries, 2020-01-23:
To bad nobody seems to respond on how to troubleshoot this.
I did additional tests and it seems that both y motors keeps spinning till both end stops are triggered then both are moving back. Can someone explain if this is normal behavior and how this system should work? I would expect that the motor would stop and that the other motor may travel max fail_distance.

jsagot, 2020-01-24:
Hi Arjan, it seems that you didn't plug the limits switch properly : they should be in parallel not in series as explained in the wiki.
Hope that would solve your issue (^_^)

ArjanDeVries, 2020-01-25:
Hi jsagot,
Thank you for replying it's appreciated!

I think i did everything right. The setup that i have is new with the latest release 1.1h see: [https://github.com/gnea/grbl/releases/tag/v1.1h.20190825].
Two motors on the y-axis and both have an end-stop so you can self-square the gantry.
One end stop is just an NC switch for the Y-axis. But the other one is in serie with Z-axis (serie is needed because they are NC switches). (see release notes)

Also the swicthes are working fine when looking at the status:
When i trigger the endstop on Motor Y the status shows Y
When i trigger the endstop on Motor A the status shows YZ

But it should be that something is wrong with the code when you have NC switches.

jsagot, 2020-01-25:
That's a nice feature !!! (but very dangerous for milling machines...)
I didn't know about this version, my bad :-)
I dig a little more and found this for the switches wiring in the define DUAL_AXIS_CONFIG_CNC_SHIELD_CLONE comment's: 
"... by sharing all limit switches on pins D9/D10[X,Y2]/[Y1,Z]"
There is no mention anywhere to plug the switches in series, but it's really unclear.
It looks like in DUAL_AXIS_MODE, you only have 2 pins for the limits D9 for X and D10 for Z AND D9+D10 for Y(Y2/D9 and Y1/D10) (Y1 should be Y axis on the shield, where Y2 should be A)

X   -> D9
Y2 -> D9
Y1 -> D10
Z   -> D10

I don't have a dual axis ready machine to try

ArjanDeVries, 2020-01-26:
Yes, it's dangerous, But i have a cheap self made cnc machine with 3d printed parts.
The machine worked fine with the previous version of grbl. I only added the new end stop. Also z homing is done first and in the second cycle the x and y. Also my config is available above.

jsagot, 2020-01-26:
That's it : try plugging the limits as i explained (A // X and Y // Z), then do the homing one axis at time (cycle_0 -> Z, cycle_1 -> X, cycle_2 -> Y) look in config.h lines 105 to 107 to configure homing cycles.
On homing you should have this behavior : 
Z axis travels to its limit on D10
X axis travels to its limit on D9
then
Y axis travel to Y1(Y -> D10) or Y2(A -> D9) or both if gantry is squared.

And then grbl should square the gantry.

ArjanDeVries, 2020-01-27:
Look at cpu_map.h also VARIABLE_SPINDLE  is not enabeld:

#define X_LIMIT_BIT      1  // Uno Digital Pin 9
#define Y_LIMIT_BIT      2  // Uno Digital Pin 10
#ifdef VARIABLE_SPINDLE // Z Limit pin and spindle enabled swapped to access hardware PWM on Pin 11.
    #define Z_LIMIT_BIT	   4 // Uno Digital Pin 12
#else
    #define Z_LIMIT_BIT    3  // Uno Digital Pin 11
  #endif

Also your text is only valid if you wanna enable variable spindle speed. In my case i don't use it

> // Variable spindle (i.e. laser mode) does NOT work with this shield as configured. While
// variable spindle technically can work with this shield, it requires too many changes for
// most user setups to accomodate. It would best be implemented by sharing all limit switches
// on pins D9/D10 (as [X1,Z]/[X2,Y] or [X,Y2]/[Y1,Z]), home each axis independently, and 
// updating lots of code to ensure everything is running correctly.

jsagot, 2020-01-27:
I was confused with the pin configuration : 
D9   -> limit X
D10 -> limit Y
D11 -> limit Z
D12 -> A.step
D13 -> A.dir
A3   -> spindle enabled
Is that correct ?
I'm even more confused with the y1/y2 switches wiring. Where do you find the configuration you used? Because in serial like you did, if both switches don't trigger at the same time, neither of them will trigger at all...
I found a compatible shield in a box (cool) and have a frame (x/y only but it should do the job) to be more helpful.

ArjanDeVries, 2020-01-27:
Yes, the pins are correct

Y2 is in serial with Z. But they are NC switches. These means that normally they are conducting. So if one of the switches are triggered the circuit will be cut and the software knows that it is triggered.
With NO switches it's different because normally they are open. So in that case you need to put them in parallel. If you hit the switch the circuit will be closed.
When you use NC switches you need to tell the software that the state is inverted.

Also i designed my own 3d printed box for my shield.

jsagot, 2020-01-27:
I almost finish the build (had trouble finding connection headers). My endstops are ramps like with 3 pins, so they should act as NO switches. I'll tell you the results.

jsagot, 2020-01-30:
So far it's working properly. I need to put the laser on it, to test for accuracy and repeatability.
(n.b : i finally found the "wiring limit switch" wiki page (^_^) )

ArjanDeVries, 2020-01-31:
After adding a lot of debugging code i solved my issue. During testing i used another CNC clone shield board. On that board i removed the cloning jumpers from the Y-axis. So that the Motor A could be moved independent. But stupid me i forgot to remove the jumpers on my machine board. So both Y-motors were just using the same steering pulses :-( I also used some jump cables to connect d12 and d13 to the stepper driver. Even that can be done more easily i found out. Just move the clone jumpers to another spot. See picture:
JumperD12D13

jsagot, 2020-02-01:
Nice for you !!!
After intensive testing it just works perfectly (it turns a crappy benbox machine into a very nice grbl one :-) ), so i'm glad you find your mess.
But there is something i do agree with you : 
#if (DUAL_AXIS_SELECT == X_AXIS) float fail_distance = (-DUAL_AXIS_HOMING_FAIL_AXIS_LENGTH_PERCENT/100.0)*settings.max_travel[Y_AXIS]; #else float fail_distance = (-DUAL_AXIS_HOMING_FAIL_AXIS_LENGTH_PERCENT/100.0)*settings.max_travel[X_AXIS]; #endif

this code is strange. I used a square config to avoid any trouble, but does it still work not on 1:1 ratio ? Maybe there is some clever calculation under the hood that lead to that kind of code but i didn't see it (or it's a simple inversion).

ArjanDeVries, 2020-02-01:
My measurements are also ok 👍 Nice that you machine is also working fine. 👍 

I thought that it was strange but it is not. It's logical. The longer the gantry (x-axis) is more movement will be possible on the y-axis with the same angle of play. So it's dependent on length of the other axis. The same axis does not make sense.

closed 2020-02-01 by ArjanDeVries

hafizipudzi89, 2021-06-26:
In the config.h i think u have to choose which type/version of cnc shield u r using e.g protoneer cnc shield V3.51 or clone one

espenjohannessen, 2022-01-16:
@ArjanDeVries Did you manage to get this working? I have a similar setup as you, and do have both jumpers in place even when using self squaring and I assume that works because duplication of signals to A is done in software and not on the board? So when I leave the jumpers ON then GRBL uses D12/13 to drive the second motor, but the jumper still needs to be in place to allow GRBL to controll the second jumper through these pins? At least that's my assumption since when I remove the jumpers nothing moves and with them on self squaring actually seems to work.

But over to the thing I'm having issues with: I've been trying to wire up limit switches in a normally-closed configuration, with two limit switches connected to Z and wired in serial, but for some reason I still can't get this to work. I have two types of limit switches available, and both have a led, capacitor and resistor on them and this seems to cause issues with the series connection between the two Y limit switch (Y and Z). Did you use similar limit switches and got it working or did you use limit switches without any circuitry on them?

https://www.amazon.com/dp/B08CDF3Q6T
https://www.amazon.com/dp/B01I1E84SG

PS! I'm using an Arduino CNC Shield V3 clone and have had everything working, but have been having some issues with noise, so that's why I'm changing over to normally-closed and was also planning on building a simple RC filter circut as described in the wiki.

https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

ArjanDeVries, 2022-01-20:
Hi,
I do have it running for a long time. I don't have the exact configuration by hand but will try to answer some questions. My switches are normally contact switches so no electronic inside. The z and one of the y are in serie. You can test the switches easily by showing the state with a command and activate them. I think you need the send the "?" command. I have also a clone board. But i can remember that there was something strange on the board. But i dont know exactly what it was. I can try to find it if you don't get it working. So start first with normal switches. Just let me know if you need help.